Denvi/Candle #527: Continuous manual mode does not work.

tistructor, 2021-12-18:
**Versions**
 - Candle version: 1.2.13b
 - Firmware version: 1.1f

**PC info**
- OS: Windows10
- OpenGL version: ...

**Describe the bug**
I added the microswitches to perform the zero machine and the hardware zeroing works but the manual axis movement mode works only in step mode and not in continuous, it returns error 15.

https://user-images.githubusercontent.com/24445373/146651267-12e2e2fa-0674-4f69-818d-50e9d89b3491.mp4

These are the parameters of the CNC3018Pro:

$1=25
$2=0
$3=6
$4=0
$5=0
$6=0
$10=2
$11=1.000
$12=0.002
$13=0
$20=1
$21=1
$22=1
$23=7
$24=25.000
$25=500.000
$26=25
$27=0.500
$30=1000
$31=0
$32=1
$100=800.000
$101=800.000
$102=800.000
$110=2000.000
$111=2000.000
$112=100.000
$120=20.000
$121=20.000
$122=20.000
$130=315.000
$131=172.000
$132=42.000
ok
 
This is the application screen:
[image]

tistructor, 2021-12-18:
Before this change it worked fine.
Will it be due to the value of the allowed ranges on the three axes ($130, $131, $132)?

tistructor, 2021-12-18:
If I run this command it works: G21G91X10F2000
This command works G21G91X-0.005Y0.000Z0.000F2000 if they are not at machine zero but it moves by 5 thousandths of a millimeter.
Continuous mode commands seem wrong!

Denvi, 2021-12-19:
Hello.
Please set $10=3 setting, as for $10=2 there is no coordinates update from firmware.
Where is your machine position reference?
Is this line uncommented in your firmware?

```
 #define HOMING_FORCE_SET_ORIGIN
```

Grab video with $10=3 in this sequence:
1) Home command $H
2) G90G0X10 command
3) G90G0X-10 command
4) Jog in continious mode X+, X-

tistructor, 2021-12-19:
Hello,
It is the native firmware of the machine I have not compiled it.
Now the X + axis seems to work continuously but Y and Z don't.
Here is the video:

https://user-images.githubusercontent.com/24445373/146671002-d7d6cac3-84f6-4d50-a647-c7b47b571fa1.mp4

[image]

tistructor, 2021-12-19:
Homing in this position:
[image]

tistructor, 2021-12-19:
I have installed this version under linux and I have no problems with jog continuously. 
candle_1.1.7.tar.gz

Denvi, 2021-12-20:
Try this version please: [Candle_iss527.zip](https://github.com/Denvi/Candle/files/7745457/Candle_iss527.zip)
 
Reference position setting added, should be like following:

[image]

tistructor, 2021-12-20:
Perfect works well.
Now I am reversing the axes to move Homing the previous position is awkward and dangerous.

tistructor, 2021-12-20:
X+, Y-, Z-

> Prova questa versione per favore: [Candle_iss527.zip](https://github.com/Denvi/Candle/files/7745457/Candle_iss527.zip)
> 
> L'impostazione della posizione di riferimento aggiunta dovrebbe essere la seguente:
> 
> [image]

tistructor, 2021-12-20:
$3=6

tistructor, 2021-12-23:
I changed some settings to have Z home with the spindle turret all the way up and again the continuous jog no longer works.

The home is the same as that of the photo with the zero of the z at the top.
x axis increases as you go to the right and y axis increases as the plane moves away.

I tried both the standard beta version and the one attached in zip file but the proplem doesn't fix.

With the stable version 1.1.7 everything works fine.

These are the settings:
$0=10
$1=25
$2=0
$3=4
$4=0
$5=0
$6=0
$10=3
$11=1.000
$12=0.002
$13=0
$20=1
$21=1
$22=1
$23=3
$24=25.000
$25=500.000
$26=25
$27=0.500
$30=1000
$31=0
$32=1
$100=800.000
$101=800.000
$102=800.000
$110=2000.000
$111=2000.000
$112=100.000
$120=20.000
$121=20.000
$122=20.000
$130=313.000
$131=174.000
$132=41.000

tistructor, 2021-12-23:
Last thing, with the beta version in the part of the window where the work plane is displayed, the tool turret with the correct coordinates is always outside the reference frame.

tistructor, 2021-12-23:
After several resets, including hardware, the jog with the standard version b does not work continuously.

With the modified B version both X and Y axes work continuously but the Z in positive does not work.

tistructor, 2021-12-23:
In modified version B, the displayed work plane is incorrect
[image]

Denvi, 2021-12-24:
Any illustrations of non-working Z-axis or wrong reference frame?

tistructor, 2021-12-25:
https://user-images.githubusercontent.com/24445373/147382855-0054699d-9125-469a-a870-a47e4a9f4588.mp4
[image]

In my view home should be above as in the real cnc, the X is correct but the Y seems to me reversed.

The video and screenshot are from the version: Candle_iss527.zip

tistructor, 2021-12-25:
I was wrong to say that the home location was like in the photo above. excuse me

Now the home is this with Z0 at the top:
[image]

tistructor, 2021-12-25:
[image]

I'm new with candle I hadn't seen these settings.
I put the Z + over the attached photo and now the Z works continuously too.

[image]

The plane displayed seems wrong to me the y should increase coming down or am I wrong?

Credo di stare sbagliando io ho impostato un sistema di coordinate che non rispetta la regola della mano destra!

tistructor, 2021-12-25:
My bad.
I fixed the problem after a few tries now it works fine.

$0=10
$1=25
$2=0
$3=6
$4=0
$5=0
$6=0
$10=3
$11=1.000
$12=0.002
$13=0
$20=1
$21=1
$22=1
$23=1
$24=25.000
$25=500.000
$26=25
$27=0.500
$30=1000
$31=0
$32=0
$100=800.000
$101=800.000
$102=800.000
$110=2000.000
$111=2000.000
$112=100.000
$120=20.000
$121=20.000
$122=20.000
$130=313.000
$131=174.000
$132=41.000

[image]

[image]

https://user-images.githubusercontent.com/24445373/147385072-38fdf362-36bd-449b-afb8-04a6fda8de58.mp4

Denvi, 2021-12-25:
I'm glad everything worked out, thank you.

closed 2022-05-07 by Denvi