Denvi/Candle #531: Candle does not run on Fedora

DilanRona, 2022-01-04:
**Versions**
 - Candle version: 1.1.7
 - Firmware version: [e.g.: 0.9j, 1.1h]

**PC info**
- OS: … Linux localhost.localdomain 5.15.10-100.fc34.x86_64 #1 SMP Fri Dec 17 14:51:10 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
- OpenGL version: ... mesa-libGL-21.1.8-3.fc34.x86_64

**Describe the bug**
./Candle 
./Candle: error while loading shared libraries: libGL.so.1: cannot open shared object file: No such file or directory

**To Reproduce**
Steps to reproduce the behavior:
1. Download linux version of Candle.
2. Unzip, to folder and go to folder
3. Execute Candle
4. Shows error when attempting to open

**Expected behavior**
Expected the software to run.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Example files**
Add G-code files that cause problems.

**Additional context**
Add any other context about the problem here.

DilanRona, 2022-01-05:
Found the problem. No where closer to solving it though.

./Candle: error while loading shared libraries: libGL.so.1: wrong ELF class: ELFCLASS64

Candle is a 32 bit program looks like, and my system is a 64 bit. I got the above error when linking the library to the libs folder of Candle.

ondras12345, 2022-01-05:
I think your best bet will be building a 64-bit binary from source.
The last time I did it, this is what was required to make it work: https://github.com/Denvi/Candle/issues/488#issuecomment-858985048