Denvi/Candle #538: Gcode from file results different from Gcode entered via command line

bobsemp, 2022-02-17:
Running Candle 1.1.7 on Windows 11 platform using Vevor 3018Pro CNC

When I enter Gcode from the command line, the CNC does exactly as specified.
When I run the very same Gcode from a file, I get an extremely small version of the same movements.

HELP.... please :-)

Bob

mar0x, 2022-03-16:
Maybe unit mode (metric/inches) is differs when you manually enter commands and replay from file.
Please share your g-code file, Grbl version, and responses for `$$` and `$G` commands.

bobsemp, 2022-03-16:
Problem was with the GRBL controller itself. Replaced the controller, problem solved

Bob

These things I have spoken to you, that in Me you may have peace. In the world you will have tribulation; but be of good cheer, I have overcome the world.” John 16:33

mar0x, 2022-03-17:
> Problem was with the GRBL controller itself. Replaced the controller, problem solved

Good to know! It makes sense to close the issue then.

closed 2022-05-07 by Denvi