Denvi/Candle #540: compiling via QT

pad73, 2022-03-11:
Not issue.
I want to adjust UI a little. You made a lot of changes from version 1.1 to 1.2. I like 1.1 design but I need scripting as in 1.2. So I want to change 1.2 to looking similar to 1.1. Mainly remove some buttons from "main" panel (fluid, freeze ....). I have problem with compiling. 3D view is black completely. Do you use some special settings for compiling?

ondras12345, 2022-03-11:
> 3D view is black completely

Service > Settings > Set to defaults

You probably have the visualizer set to black lines on black background.

pad73, 2022-03-11:
Perfect help, thank you, works great. Now I can do what I need :)

pad73, 2022-03-13:
Finally I found last source code is version 1.2b, so some small differences from 1.1 version, but without scripting possibility. Does anybody knows where to take source for last version with scripting?

mar0x, 2022-03-13:
@pad73
I believe 1.2 version is developed in '_Experimental_' branch of main repository (see https://github.com/Denvi/Candle/tree/Experimental).

closed 2022-05-07 by Denvi