Denvi/Candle #541: Abort and Pause (under the gcode window) in Candle 1.2

pad73, 2022-03-15:
**Versions**
 - Candle version: 1.2.15
 - Firmware version: 1.1h

I found in Candle 1.2 some different behaviour of the buttons Abort and Pause. In Candle 1.1. after click on the button action was immeditatelly, when click on Pause then program stop (HOLD), when click on Abort program stop and terminate immeditatelly.
BUT
In 1.2 (15) version after clicking on both these buttons program continue with actually loaded data and goes to the pause or stop after moving by these last datas. It means there is long time between click and action. It is very uncomfortable and may be, little dangerous. Only way with pause is to use HOLD button (that is OK, it works perfectly). But to terminate program imediatelly the way is only one, RESET. Not so good because it means loosing working coordinates. Of course you can put it back using stored coordinates using G92. But as I wrote before, it is very uncomfortable.

And I found that reaction to click on moving buttons (x y z) is slower than Candle 1.1

Can anybody explain it? Is it any settings to change buttons behaviour?

Now, when I´m writing this text I have idea to use shortcuts because there is possibility to add shortcut to Hold and Abort, but I can test it just now. Otherwise if it will work I will use it, but original buttons funciton will stay strange to me.

pad73, 2022-03-16:
Yesterday I tested function of Pause and Abort using shortcuts ... works the same as when using buttons. This is not acceptable for me so I´m leaving 1.2 version, and I´m going back to 1.1. This 1.2 version brings some new nice tools (scripts, working area borders, ...), but changed more important thing to unusable. 
If anybody has solution let me know please.

rjcamatos, 2022-03-16:
O think it is something about cache cuue, you need to wait until last cache
finish then it will pause!
To emergency stop push button cancel.

pad73, 2022-03-16:
You wrote nothing new, but how to solve it? It seems to me the base code for Candle 1.2 was reprogramed to use some new functions and it brings these issues.

mar0x, 2022-03-16:
> Can anybody explain it?

According to source code, the "Pause" button in "[Experimental](https://github.com/Denvi/Candle/blob/1c29372b4ea9940370a4b73d6fa5a2232c058013/src/candle/frmmain.cpp#L743)" branch (v 1.2) changes the _sender_ state i.e. affects only commands from the file to send. The commands which was sent are processed by CNC as usual. This why you experience the _delay_ before CNC stops.
As far as I understand, the "Hold" button was introduced exactly to immediately pause the CNC ([source](https://github.com/Denvi/Candle/blob/1c29372b4ea9940370a4b73d6fa5a2232c058013/src/candle/frmmain.cpp#L861)).

In [stable](https://github.com/Denvi/Candle/blob/3f763bcde1195e23ba119a5b3c70d7c889881019/src/frmmain.cpp#L2473) version (v 1.1), the "Pause" send to CNC "!" character processed by CNC right as "Hold" button in v 1.2.

The "Abort" button in "[Experimental](https://github.com/Denvi/Candle/blob/1c29372b4ea9940370a4b73d6fa5a2232c058013/src/candle/frmmain.cpp#L762)" branch (v 1.2) just enqueue "M2" command. So CNC will process the commands which was already sent and then stop. Again, with delay.
The "Reset" button in "[Experimental](https://github.com/Denvi/Candle/blob/1c29372b4ea9940370a4b73d6fa5a2232c058013/src/candle/frmmain.cpp#L850)" branch (v 1.2) does the actions similar to "Abort" in [stable](https://github.com/Denvi/Candle/blob/3f763bcde1195e23ba119a5b3c70d7c889881019/src/frmmain.cpp#L1982) version.

>  Is it any settings to change buttons behaviour?

I can't see in source code the way to change it's behaviour. Just use "Hold" button to immediately pause CNC.
Regarding "Reset" behaviour difference, could you confirm the difference in v1.1 "Abort" and v1.2 "Reset" button result?

pad73, 2022-03-16:
Thank you for explaining. Hold is Ok, there isn't problem. But to use reset as abort program isn't the best way IMHO.

mar0x, 2022-03-16:
"Reset" in v1.2 does exactly the same as "Abort" in v1.1: send "[soft-reset](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Commands#ascii-realtime-command-descriptions)" command. I've tested - machine and work coordinates remains unchanged. Could you please confirm it?

This is the question to @Denvi what is the purpose of new "Pause" and "Abort" in v1.2 ?

pad73, 2022-03-17:
Thank for your answer. I will test it evening. I used it yet but it looks strange to me. But OK, I will test it again focused on.

pad73, 2022-03-17:
Sorry for my English 😉 https://youtu.be/mM8WeGjbDXs

mar0x, 2022-03-17:
Thank you for video! The difference is obvious.

I'm sorry, I was wrong about equivalence of "Abort" in 1.1 and "Reset" in 1.2. This is not true.

"Abort" in 1.1 immediately send "`!`" (which is the same as "Pause" in 1.1 or "Hold" in 1.2) and `m_aborting` flag raised.
The only way to send "`!`" in 1.2 is "Hold" button.

"Reset" button in "Control" panel should behave same way in 1.1 and 1.2.

pad73, 2022-03-17:
Result of it is that isn't cool ... It is quite stupid

mar0x, 2022-03-17:
Sad but true.
Consider following sequence for "Abort":
1. "Halt" for immediate stop the machine (remember work coordinates to check);
2. "Reset" (in "Control" panel) to stop processing file;
3. "Restore Origin" to restore work coordinates (check the restored coordinates from the step 1);
Agree, this is not as simple as one button click, but should work.

Maybe @Denvi can suggest something different...

pad73, 2022-03-18:
Easy way  - let it works as before 😉

SinkaGit, 2024-03-01:
Could you please fix this?