Denvi/Candle #544: Height map always probes origin.

cirvladimir, 2022-03-20:
**Versions**
 - Candle version: 1.1.7
 - Firmware version: 

**PC info**
- OS: Windows
- OpenGL version:

**Describe the bug**
I created a heightmap grid with borders that don't include the origin. I expected only points within the grid to be probed. However, candle always starts the probing process by probing the origin. This is an issue because I had a clamp at that point.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a heightmap with X: 10, Y:10, W: 10, H: 10, grid of 2x2.
2. Click on probe

**Expected behavior**
I expect the tool to move to 10,10 and then probe. Instead it moves to 0,0 and probes, and then moves on to probing the grid.

pad73, 2022-03-20:
Candle takes reference height on origin. Take your care about it and don´t place clamp on origin. Easy solution.

ondras12345, 2022-03-20:
Yes, that's an easy solution, but it also complicates the process of making a PCB. I usually put the origin in the bottom left corner of the PCB, so that I can use two existing edges of the material and don't have to cut out the whole outline.
I think it would be a major improvement if it probed a corner of the heightmap grid instead of origin. It shouldn't be all that hard to implement, either.
A similar issue has been reported before: https://github.com/Denvi/Candle/issues/357

pad73, 2022-03-20:
It is very easy to adjust probe area. Don't use Auto mode to define area but resize it by yourself changing area size and start point place. I made easy video to explain what I mean. Sorry for my poor English. I hope it will help to you. https://youtu.be/0Od6XLDEWMM

ondras12345, 2022-03-20:
Yes, that works really well if your origin is in the middle of the probe area. But if you put the origin eg. in the bottom left corner of the board outline in CAD and generate the Gcode from that, your probe area will start at say X10 Y10, but Candle will still need to probe X0Y0 before it moves to the probe area.
Honestly, it never occurred to me that I could just set the origin to the middle of the board in CAD. I'm not sure how well it would work with my FlatCAM workflow, but it seems like it should be doable.
However, I still think that Candle should probe the center or one of the corners of the probe area instead of origin. It should work with any Gcode file, the user shouldn't need to generate it specifically to work around this bug.

pad73, 2022-03-20:
You're right. First touch is in x0y0. But I think changing start position is the easiest way to solve your troubles. And to have start position not exactly in any corner isn't dangerous or unusable. Only you need to learn it. I believe you will find it can be very useful to start not only in the corner 😉.

rjcamatos, 2022-03-22:
I, i'm using candle 1.2.
When i do a height map.
For example for G56 then milling processo works well.
Then i Change for example for G57 do ano height map select the height map
check big afer loading the correct One and the milling GOES wrong.
It's look like it takes the first prove of G56 as reference. Of course on
G56 I do ano height probe as well.
How can i correct the issue?
This is for CNC a PCB board of 1Oz.

ondras12345, 2022-03-27:
So I tried putting the origin in the middle of the board today. No problems in FlatCAM, but I encountered some on the CNC.
If the origin is not exactly in the center, it makes it quite hard to set up the machine so that two existing edges of the board are used.
It also makes it hard to probe the heightmap after something has already been milled, because the copper layer may no longer be conductive all the way to the center.
It is a nice workaround and I will probably use it from now on, but it's still just a workaround. Candle shouldn't probe at 0,0 in the first place.

FrankHovis, 2023-05-21:
Bumping this unexpected behaviour after snapping off the end of an tool due to the unexpected probe attempt at 0,0. The probes should always be in the border area.

pka4916, 2025-08-21:
i agree..  having the problem everytime..    this really needs to be fixed.