Denvi/Candle #545: Candle windows 11

denhoy81, 2022-04-05:
Does Candle work on windows 11?

mar0x, 2022-04-07:
Yes, it does.

closed 2022-05-07 by Denvi

lrafuse1005, 2023-10-15:
i am having a terrible time making this work   i think i have tried everything but i am new to this   it starts, mills for a minute then stops   works longer everytime but never in the same spot

HELP!

mar0x, 2023-10-16:
Hello,
To advise anything useful it would be good to know the details about your CNC (model, firmware, etc).

For now, I guess you are running Windows 11. One of the possible reason of the behavior you have described is unreliable CNC control board power or USB connection.

lrafuse1005, 2023-10-17:
Thank you for replying!  I have a vevor 3018 prover

lrafuse1005, 2023-10-17:
I read some more into some possible issues and decreased the windows running system to run windows eight for this particular product. It states it can run at 11, 10… But I am having trouble with issues that it is stopping. I will try this morning and see how it runs on windows eight and, let you know how I do. Thank you so much for your quick reply. I honestly didn’t know if anyone would reply since this thread is so outdated.