Denvi/Candle #546: height map not working for me

ColAstro, 2022-04-06:
I have two versions of Candle. In the older one the z-probe works fine. In v1.1.17 I had to change settings
touchCommand="G21G91G38.2Z-30F100; G0Z1; G38.2Z-2F10"
to
touchCommand=G90G21G38.2Z-50F100\nG92 Z13.88    \nG0Z28    \n
To get the probe to work but when I use heightmap it doesn't detect when the probe touches the surface. 
How do I fix this?

Thanks

mar0x, 2022-04-07:
Hello.

What is the reason of `G92 Z13.88` command for manual probe?
What is the settings of "Z top" and "Z bottom" for Heightmap probing?
Does the CNC continue to move probe down after touch?

Please enable "Show G-code program command" in "Console" section and share the console output during manual and heightmap probe.

RafaBRxx, 2024-09-10:
I have the same problem,  the probe works well, but on mapping all results are 0.000 ...  GRBL Ver 1.1h.20190830
[image]
[image]

mar0x, 2024-09-11:
@RafaBRxx As you can see from console output you CNC reports 5 coordinates for probe point. While Candle [expect only 3 values](https://github.com/Denvi/Candle/blob/3f763bcde1195e23ba119a5b3c70d7c889881019/src/frmmain.cpp#L1174). Correct line is:
```
   QRegExp rx(".*PRB:([^,]*),([^,]*),([^:,]*)");
```

papillion47, 2024-10-05:
Heightmap, visualizer ve jog komutları çalışmıyor. Candle yükledim, aynı. GRBL yi defalarca silip yeniden yükledim, aynı.. Lütfen bir çare !.. Eski bilgisayarımda da aynı grbl yüklü, orada bir sorun yok. Yeni ve hızlı çalışan bilgisayarımda başta normal ve doğru çalışıyordu. Sonra ne oldu bilmiyorum, bu sorun ortaya çıktı.. Gidermek için ne yapmalıyım? Lütfen biri yardım etsin.. Selâmlar..

papillion47, 2024-10-05:
[image]
[image]
Heightmap, visualizer ve jog komutları çalışmıyor. Candle yükledim, aynı. GRBL yi defalarca silip yeniden yükledim, aynı.. Lütfen bir çare .. Eski bilgisayarımda da aynı grbl yüklü, orada bir sorun yok. Yeni ve hızlı çalışan bilgisayarımda başta normal ve doğru çalışıyordu. Sonra ne oldu bilmiyorum, bu sorun ortaya çıktı.. Gidermek için ne yapmalıyım? Lütfen biri yardım etsin.. Selâmlar.

mar0x, 2024-10-05:
@papillion47 try [this](https://github.com/Denvi/Candle/issues/new?assignees=&labels=&projects=&template=bug_report.md&title=)

papillion47, 2024-10-07:
Denememi istediklerinizi becerecek kadar PC bilgim yok. Sanırım birileri bana "uzaktan yardım" programıyla yardımcı olursa ancak sorunum çözülecek. Hayırsever ve PC bilgileri kuvvetli biri çıkarsa mutlu olacağım. Teşekkürler..

mar0x, 2024-10-07:
@papillion47 As I [wrote](https://github.com/gnea/grbl/issues/95#issuecomment-2394338109) before, to give you an advise it is required to know your environment (at least). New issue template is a basic list of questions you will need to answer.

If you feel your PC experience is not enough, it's a good chance to improve it by systematically describing, understanding, and solving the real issue.

To use latest Candle versions (most likely) you will need to upgrade your CNC GRBL firmware and this requires an experience and patience. Otherwise, you need to stay on old "Candle" or "GRBL Control" versions.

papillion47, 2024-10-10:
mANY THANKS..

Hostaler, 2024-10-13:
> ![penceresi de grbl](https://private-user-images.githubusercontent.com/69798943/373861999-b067f3e9-23d3-4a56-bca2-e88c6495fab0.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjgxMjUwOTEsIm5iZiI6MTcyODEyNDc5MSwicGF0aCI6Ii82OTc5ODk0My8zNzM4NjE5OTktYjA2N2YzZTktMjNkMy00YTU2LWJjYTItZTg4YzY0OTVmYWIwLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMDUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDA1VDEwMzk1MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWJhMGYzNjFlMTUwODY5YTIyZDc4NjVjMDI1NzVhMTY2Mzk0MmZjNDE1MGUyM2JmMTc3OTMyOWM2ZDgxZDdiZjAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.gUp0r3yUphA4nhGAMgPueumvBRNpG-IGuoVmhmXe5eM) ![El programa de entrenamiento de grbljog](https://private-user-images.githubusercontent.com/69798943/373862003-022453ec-60dd-4711-836f-bcac64c2531d.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjgxMjUwOTEsIm5iZiI6MTcyODEyNDc5MSwicGF0aCI6Ii82OTc5ODk0My8zNzM4NjIwMDMtMDIyNDUzZWMtNjBkZC00NzExLTgzNmYtYmNhYzY0YzI1MzFkLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMDUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDA1VDEwMzk1MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTAzYjg0MmRiNDFkNjQwMThhZjM1MmYxODE0MWU0N2E2ZDlkODExZTg4YzQ1OGIxNjdkNTUyYzNjMjNiZTEyZDEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.SvuP1J4XUkcBmOBqQ8RZNhVbs2ukM8uAAfibrKJIjiw) Mapa de altura, visualizador y jog komutları çalışmıyor. Vela yükledim, aynı. GRBL yi defalarca silip yeniden yükledim, aynı.. Lütfen bir çare .. Eski bilgisayarımda da aynı grbl yüklü, orada bir sorun yok. Yeni ve hızlı çalışan bilgisayarımda başta normal ve doğru çalışıyordu. Sonra ne oldu bilmiyorum, bu sorun ortaya çıktı.. Gidermek için ne yapmalıyım? Lütfen biri yardım etsin.. Selâmlar.

 why don't you write in English or Spanish, which are two of the most spoken languages in the world.