Denvi/Candle #549: Users command popups

Jarewa, 2022-04-17:
Candle 1.2.14b 

I wanted to ask if there are other scripts for pop-ups, I would like to insert text and to confirm or deny that it executes depending on what I do?

{var x = QInputDialog.getText(0, "Relative", "X")}

This script causes a window to open and you have to enter the data to be captured and inserted

G91 G0 X{x} 

I would like to ask if there are other references to other pop-ups to create my own scripts?

Jarewa, 2022-04-21:
I tested a bit and some qt5 windows work "QInputDialog" "QMessageBox"  

But it is limited in the number of characters. Maybe the author would allocate more space?

{QMessageBox.warning(this,"Ostrzeżenie","Wymień narzędzie i naciśnij Ok", QMessageBox.Yes | QMessageBox.No )}

Did not work  "QMessageBox.Yes | QMessageBox.No" :(

Denvi, 2022-04-22:
Hello, im tried to remove limit, but it's may be buggy for now:
[Candle_iss549.zip](https://github.com/Denvi/Candle/files/8542608/Candle_iss549.zip)

This script should work:

```
{script.importExtension("qt.widgets")}
{with (QMessageBox) warning(0, "1", "2", StandardButtons(StandardButton(Yes), StandardButton(No)))}
```

Jarewa, 2022-04-23:
{script.importExtension("qt.widgets")}
{with (QMessageBox) warning(0, "1", "2", StandardButtons(Yes, No))}

Works I'm happy, I already know how to format other windows as well :)

Can you still tell me if they work and how to format the conditional statements?

np. 

if( x== yes){
  // code
}
if(x== No){
 // code
}

Denvi, 2022-04-25:
Some fixes for nested brackets:
[Candle_iss549_1.zip](https://github.com/Denvi/Candle/files/8556356/Candle_iss549_1.zip)
 
So now this example works:

```
{script.importExtension("qt.widgets")}
{with (QMessageBox) var r  = warning(0, "1", "2", StandardButtons(Yes, No, Cancel))}
{if (r == QMessageBox.No) {s = "G"; s + 0} else {"G1"}}
```

Jarewa, 2022-04-26:
Thank you very much, it gives you a lot of possibilities and is very functional.

Now tests :)

mar0x, 2022-04-26:
@Jarewa Maybe it worth to write you own plugin instead of user command scripting?

davi-domo, 2022-05-26:
hello, 
translation via google translate

thank you for this feature
I encounter a problem, I make user commands for probes
cutter diameter dialog box: OK
on the other hand I cannot add numbers, it treats them as strings
on the other hand if I work in the negative it works

this works
`{script.importExtension("qt.widgets")}
{d = QInputDialog.getText(0, "Relative", "Diamètre de la fraise")}
{y_probe = 25}
G91
G21
G91 G0 Y{-y_probe -d}`
return
`
G91 < ok
G21 < ok
G91 G0 Y-27.0000 < ok`

this does not work
`{script.importExtension("qt.widgets")}
{d = QInputDialog.getText(0, "Relative", "Diamètre de la fraise")}
{y_probe = 25}
G91
G21
G91 G0 Y-{y_probe +d}`
return
`G91 < ok
G21 < ok
G91 G0 Y-252 < ok`

I tried to process them in integer
but it returns me a valid error

davi-domo, 2022-05-26:
> Hi, have you tried parseInt() or unary "+" such as:
> 
> ```
> y_probe + parseInt(d)
> or
> y_probe + +d
> ```

thank you
I tried but I must have made a typo
I will be able to advance on my probes

[image]

JepApps, 2023-02-12:
> Some fixes for nested brackets: [Candle_iss549_1.zip](https://github.com/Denvi/Candle/files/8556356/Candle_iss549_1.zip)
>
> So now this example works:
>
> ```
> {script.importExtension("qt.widgets")}
> {with [...]

Hello i tried the following command to generate a confirm pop up, but i have the following error "SYNTAXERROR: PARSE ERROR ELSE G1} < error:2", have u found some command to generate a confirm dialog?
- Candle version "1.2.15b"
Thanks 👍

Jarewa, 2023-02-12:
Something wrong here, show your script?

JepApps, 2023-02-12:
Here its the script
[image]

JepApps, 2023-02-12:
> Something wrong here, show your script?

It's the script 
[image]

closed 2025-09-08 by github-actions[bot]