Denvi/Candle #550: Height map irregularities

104TMR, 2022-04-24:
Hi Denvi,
Many thanks for a great piece of software.
I'm using Candle 1.1.7 with a Vevor 3018 CNC mill to mill PCBs.

I undertake a height map before any PCB mill, and tick the box in Candle to use the map, but I'm finding that the height map doesn't seem to be providing the height corrections that it should.  The cutting tip (0.1mm, 30deg) cuts down through the copper layer in some areas, but then only cuts part-the-way-through, or not at all, in others.

I decided to check the consistency of the height map process, and ran 10 consecutive height maps over exactly the same area (10 point x 10 point, height map probe feed =10), and looked at the .map files these runs created in Excel.

I've been concerned to find some dramatic and variable differences between the (what-should-be identical) height map coordinates.

I selected the first (of 10) height map as a reference, and then subtracted each subsequent height map array from the reference, and looked at the resulted difference arrays.

These different height map arrays showed fluctuating differences, both in their magnitudes and distribution.  While most deviations fell in the -30 micron to +30 micron range (still surprisingly large, considering that the copper layer on a 1 oz PCB is 35 micron), several difference maps showed some points that were 200 to 900 micron deviant(!).

These latter figures are clearly in error, but it seems they all might be, as no two deviation arrays were the same.  I would have expected to see typical machine and randomisation errors of 0~5 micron, perhaps, but not 20-30 micron, and certainly not 200-900 micron.

Could I ask your comment on what you think may be happening here?
I can send the Excel file with the height map difference arrays, if it's useful.

Many thanks for your help,
Glen

Jarewa, 2022-04-24:
Increase step accuracy and decrease point homing speed.

Because how many steps do you have per rotation?

104TMR, 2022-04-24:
Hi Jarewa,
Thanks for your advice.

How do I increase step accuracy, and where do I find the steps/rotation?

The point homing speed is (default) 10(mm/sec?), which seems fairly slow
already.
Do you have a recommendation that typically works well?
Thx…

Jarewa, 2022-04-25:
Send the command $$ and write what you have under the items $100 $101 $102 

10(mm/sec) the slower the better

mar0x, 2022-04-25:
Hello,
The **F**eed rate is specified in "_units_ per **minute**". For **G21** it is "mm per minute".

@104TMR , I would recommend to check the accuracy of the probe process itself. Repeat the following command sequence in console:
```
G21 G91 G38.2 Z-40 F10  ( do the probe and check the last number in response [PRB:0.000,0.000,-1.031:1] Z=-1.031 )
G21 G91 G0 Z1 ( move the spindle 1mm up )
```
Check the dispersion of Z probe response. If it is stable enough, then it makes sense to continue your investigation.
( I've got -0.011 - 0.013 for first 5 probes, than it changes to -0.028 - 0.031 for latter 6. But I'm using V-cut as a probe for copper PCB and each probe may push in a copper layer a bit. )

The probe process is sensitive to feed rate: the lower feed rate the more accurate result. Candle 1.1.7 does not use "_Heightmap probing feed_" setting value at all. The last movement feed rate is used. So, try manually execute the sequence above right before the heightmap probe and control the feed rate in a left-top corner. Consider [unofficial v1.1.9](https://github.com/mar0x/Candle/releases/tag/v1.1.9) with fixed "_Heightmap probing feed_" setting usage and reasonable small feed rate configured (10 - 30 mm/min).

104TMR, 2022-04-25:
$101=$102=$103=800.
The probe feed rate is at 10mm/sec.

However, I suspect I may have found a contributor to the problem: the
polarity of the probe electrical connections (to pins A5) were reversed
from what I had thought I’d connected them: positive was on the work piece
and negative was on the probe.

I decided to check the electrical signals when I noticed that one of the
height map probe excursions only went part-way down towards the PCB
surface, but then pulled back and recorded (an incorrectly high z-value),
without actually having reached the copper surface.

Clearly, the A2 pins had received some (bogus) electrical signal that told
the controller that the probe had reached the surface, when in physical
reality it had not.

So I put an oscilloscope on the A2 pins to try and see what electrical
behaviour might be occurring to cause this, and then noticed the reverse
polarities.

I’ve since reversed the polarities of the A2 connections to the probe and
work piece, and run another set of 10 identical height maps, and found that
the unrealistic (200-900 micron) differences between them are not present
anymore.

I am seeing maximum differences between height maps of 10~20 micron, which
is still a bit concerning, compared to the 35 micron thickness of the
copper, but nothing like the outliers seen previously.

I've also just tried milling a PCB using a new height map on a new board,
but the results are not looking good.
Please see attached pic.  Shows severe undercutting and overcutting in
various areas.

Further thoughts appreciated.

104TMR, 2022-04-25:
Thanks for your advice, Max.
You will see I have just posted an update, describing the polarity reversal
issue I discovered.  But there still seems to be some significant height
map deviations.
I'll investigate your suggestions tomorrow and will advise.

104TMR, 2022-04-26:
Hi Max, I tested your two lines of code, and got z-depth readings that stayed within 1 micron of the 1st reading over a set of 10 readings - which is a great result!
Question still remains: Why does a set of repeated, identical height maps produce differences of up to ~20 micron?
However, I've done some reading, and noted that it looks like the max. steps/mm for the stepper motors can be up to 1600.  So, I've set z-steps/mm to 1600 ($102=1600, leaving $100=$101=800) and rerun the multiple (10) height maps.
Remarkably, the array of differences of each height map from the average of all height maps are showing maximum and minimum values of less than 7 um, which is 3x better than what I was achieving previously.  I have since done a new height map and run a PCB routing run using the new steps/mm, and the cutting depth consistency is much better.
Proof of the pudding...

mar0x, 2022-04-26:
> got z-depth readings that stayed within 1 micron

So, the probe itself works fine **when the probe runs at 10 mm/min feed rate**. Perfect!

Have you checked the actual feed rate during height map producing (at the top left corner, next to **F/S**) ?

Grbl $102 setting is the number of steps for stepper motor to move spindle for 1 mm along Z axis ([source](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration)). It is conditioned by your stepper motor (number of steps per rotation) and mechanics of you machine. If you have 800 steps/mm and your Z-axis movement is 100% accurate it is incorrect to set $102=1600 just because your Z axis starts moving twice longer (in mm). This is it. There will be no double precision etc. And you've got better probe results just because movements along Z-axis are doubled. Use the ruler and measure the height after "G21 G91 G0 Z20" command (there should 20mm difference).

104TMR, 2022-04-27:
Yes, I see what you're saying, Max.  Indeed, setting $102=1600 makes the z-displacement 2x what it should be.
Clearly, I'm a novice with all of this.
So, I've reset $102=800, set the probe speed to 2 mm/s (and Yes, F/S reads 2 (or whatever speed I set it to in the Height Map Settings, so it does seem to get it's value from there)), and rerun 10 height maps, and find much the same as previously: differences of +/- ~17 um.
So, why can the machine get such repeatable precisions of 1~2 um in a single test mode  (using G21 G91 G38.2 Z-40 F10, etc.), but such lousy precisions when doing a height map (even with a very slow probe speed of 2 mm/s)?

104TMR, 2022-04-27:
Sorry, "12 um" should be "1~2 um"...

chfgwd, 2022-04-27:
I'm using Lightburn 1.1.03, when I have a layer set to line to outline
shapes or text, the laser does not turn off during "Traversal Moves". Is
there a setting in the "$" commands that will fix this?

104TMR, 2022-04-28:
I've been doing a number of height map tests under different conditions, but haven't got to the bottom of why z-differences are so small when measured at one point, and why they start varying so much when tested between varying x-y positions.
Following are some descriptions and plots of the results.  Interpretations to follow.
The height map scans were a 50x50mm square of 4 points (ie. each corner).  The 4 depths recorded at each corner were averaged over the 10 scans, and then each of the 4 points for each scan were subtracted from the average (to create differences from the mean), and the minimum and maximum values for each scan plotted vs scan number.
1.  I wondered if the z-probe trigger signal may be affected by RFI from the z-stepper motor (I notice that the nearby AM radio gets heavy RFI when the stepper motors run, so I put a 0.22 uF capacitor across the A5 terminals, to dampen any RFI that could affect it.  Didn't make any difference.
[image]
Image shows maximum and minimum deviations from the mean for 10 scans.
Shows min/max ranges of approx. +/- 15 um.  
What does stand out from the plot is the apparently strong correlation between max and min values across the scans.  
This seems to imply that when the maximums are going up, so are the minimums, and vice-versa.  
If we were dealing with a purely random effect causing the variations, we would not expect to see correlation between the max and mins - instead random effects would make the values independent of each other across the scans (disagree if you like...!).
2.  To check this further, I disconnected the capacitor, and ran three more 2x2 scans of 10 scans each (ie. 30 scans in total), to see if the max/min correlations continued.  It looks like they do:
[image]
[image]
[image]
3.  So then I thought that perhaps a systematic offset may be associated with one axis drive (X or Y).  So I did a set of 10 height scans between two points on the X-axis, and then the same for the Y-axis.  Plots below show the max/min differences vs scan number, and the correlation seems to have disappeared:
[image]
[image]
4.  I then did several height tests at single points (ie. not moving the X/Y position), and found that the height differences reduced to 1~3 um, as found previously.
This seems to be showing that when you don't move the X/Y dimensions, height values are very precise.
However, as soon as you start moving around in the X/Y dimensions, variations creep in, and make the height map fairly unreliable.
I don't know what else to interpret from all this.
Further thoughts appreciated from anyone interested in this topic.
Does anyone else experience PCB isolation routing showing unreliable depth cutting - or does everyone else's projects work perfectly and mine's the only dud?! ;-)
Thanks for any thoughts...

Jarewa, 2022-04-29:
What screw do you have, TR8x2?

104TMR, 2022-04-29:
Not sure.  Their outside diameter is 8mm (hence the "8" in "TR8"?), and
they move the head 4 mm with 1 turn.

Madbyte3d, 2022-05-29:
104TRM,
I think that you are not alone. I've been pulling my hair for a week or so because of inconsistencies in isolation routing. My case is a bit different in that sometimes I get really deep mills, and therefore wider mills.  For a while the tip was barely touching the pcb and just engraving it. I made a very basic circuit to repeat the test.
[image]

In my case all the isolation maps increase in height as the probing progresses from bottom left to top right. This means that my spoilboard is not level but the thing is that I just surfaced it with a 1 inch surfacing bit.

Do you mind sharing the height map parameters that go with the charts you shared above? Also the excel you are using for graphing? I'll attempt to run the same tests generate the graphs and report back. 

I'm new to CNCs but have several years of experience with 3d printers so I understand the mechanics of the steppers motors and controllers.  Is a different mindset when it comes to the relationship between the tip of the bit and the surface and having two sets of coordinates to deal with.

I have a Sainsmart Genmitsu 3020 Pro Max if that's important to anyone.

scorninpc, 2022-08-31:
Hello guys!

How are you going with this problem?

I've tested all in this issue, and recreate all the tests, and got the same results with no fix

So i decided to add 1/32 microsteps, and got some better results, but no success yet

So i tried to link my oscilloscope to see whats happening, i saw alot noise. 

I added a pull up with divider to trigger SCL, creating a route for noise escape

[image]

[image]

and i got a nice result

3 times sequentially of the same point

[image]
[image]
[image]

This occured on 8 tries, but sometimes keep falling

[image]

but it's reduced by 80~90% of errors

Maybe it is a direction. I need work better with this resistor values

104TMR, 2022-08-31:
Hi Bruno,
Good on you for persevering with this, and it looks like you may have a
good technique for improving the height map performance.

Personally, I have moved away from routing, and instead mounted a 40W laser
on the router head, and am now spray painting the blank PCB, letting it
dry, then cutting through the paint layer with the laser, then drilling the
through holes with the router & chuck, and then etching with H2O2 + HCl,
then dissolving the paint off with acetone, and the PCB is done.

Works very well.

I might try routing again at some stage, but for now the laser/paint method
allows me to produce the boards I want without too much hassle, so I’ll
stick with it for now.

Cheers,
Glen

scorninpc, 2022-09-01:
Ok for probe tests, but no acceptable on map height =/

[image]

pixelwaster, 2022-09-01:
Here is a video I found interesting, A quick demonstration on how much overshoot happens when you use continuity based probing.

From what I understand, the overshoot is a result of the feedrate and stepper acceleration, The faster the acceleration and slower the feed, the less overshoot. 

[Stop probing with your tool  ](https://www.youtube.com/watch?v=xBeWwSBjpAo)

scorninpc, 2022-09-02:
Nice video. But i think he show the precision of probe, not the different results of the same points (you can see he moving the plate with hands alot time)

Anyway, today i'll do some tests with acceleration, becase i done alot with speed, bot not with accel

Thank you

scorninpc, 2022-09-16:
I found a loooooonnng discution of that https://github.com/gnea/grbl/issues/96

And the summary: https://github.com/gnea/grbl/wiki/Wiring-Limit-Switches

I ordered the optos, and will test with probe as i can, and post results

104TMR, 2022-09-16:
> I found a loooooonnng discution of that gnea/grbl#96

Hi Bruno,
Interesting info on limit switches, although I’m not sure how this relates
to the topic of this thread - ie. height map irregularities?
Cheers…

scorninpc, 2022-09-19:
Sorry, but i think you dont understand your first problem, and dont read the related links. 

All this problem is causes by noise and the links is about cut that, isolating contacts with opto-isolation

The height map irregularities is not caused by software problem, but by measurement

104TMR, 2022-09-19:
Hmmm…ok, I’ll look forward to seeing what the eventual solution is.
Thx…

Mot007, 2023-08-08:
I have been reading and following your comments re the height map inconsistencies. I recently bought a 3018 pro and also tried to cut a simple circuit PCB 108x30 cm.
I have almost broken every bit In the process to try and auto levelI. I am a retired hardware engineer. Sometimes the hardware is not what it should be and the battle of the software versus the hardware is still evident! I put a scope on the input if the Z probe and to my amazement saw quite a lot of bounce. The problem as I see it is on the second reading and again I'm not sure what the software is doing.
But it seems that the first plunge gets a clean low upon touching the copper. Then the probe rises and descends to get a more accurate height reading. This is where lots of bounce occurs. As the smaller steps just kiss the copper, lots of high resistance connections appear causing irregular connections untill rock bottom connection occurs. 
A simple solution would be to step down, denounce for a given time, read, and repeat. Another solution would be to increase the pull up on the sense input to much greater than  10k maybe even 100k . I have captures of this but don't know how to share it with you. I'm happy to do so if you show me the way.🙂

Mot007, 2023-08-08:
[image]
[image]
[image]

Mot007, 2023-08-08:
Ok I missed the obvious share photo!
The first image is the first plunge, 2nd and 3rd is on the final smaller incremental steps to Zero.

104TMR, 2023-08-11:
Hi MOT007,
Those are interesting results you’ve got for the height probe resistance
measurements, and Yes, likely that it has something to do with the
uncertainties that are throwing the calibrations off.

However, as mentioned in an earlier post, I have found the laser scribing
of painted PCB (and H2O2 +HCl etchant afterwards) to be an accurate and
reliable method for creating the PCBs that I bought the 3018 to do.
So I haven’t been pursuing further analysis of the variable height-mapping
problem.

Your results however, do seem to indicate that the software probably could
be modded somewhat to try and correct the effects you’ve identified.
However, I’m not sure who would do that - unless your into playing with the
source code yourself?

Cheers…

Mot007, 2023-08-11:
Hi TMR104, 
I have gone along the path of a hardware fix to the noise and have not made much of a difference. Today I tried a few different ideas and have had a much better result in the deviation for the same map. I have proven to be definitely sure that a Shap tip will increment the error on every read. This is due to the initial velocity to make contact and also the subsequent slow plunge makes a difference. Coupled together with a flat probe will reduce the error for the same point .
This all fell to bits once I repeated a simple height map (30x30mm) 3,3 probes.
What I found is that the plunge is F10 and could not find where to alter this. More importantly the inaccuracy deviation seems to show much greater as the machine scans the centre points traveling right to left. ? Much more investigating I guess.
Are you able to point me to the software? I could only  have a go but not qualified in that department. 
On the other way of producing, what is the minimum track/space that you can get which Lazer?
I only have a 2.5W one and have not fired it up as yet.

104TMR, 2023-08-11:
Hi MOT007,
It sounds like there's some persistent gremlins in these 3018's ...
Re. laser etching the painted PCB, I usually make my tracks 0.5mm with
0.5mm spacings, but have had no trouble getting 0.3mm tracks with 0.15mm
spacing on projects that need this.
I think a 2.5W laser might not have the grunt needed to burn through the
paint at a reasonable rate.
I've got a 40W laser, and I power this along at 600mm/min, and it does
well.  Of course, it's a cheap version, so I'm pretty sure the real power
out of it won't be 40W, but it still seems to do OK.
Mine is similar to the following on eBay:
https://www.ebay.com.au/itm/265345048679?hash=item3dc7cc2067:g:2H8AAOSwfnNh382-&amdata=enc%3AAQAIAAAA8F1BFjV%2BgCtg5zV7ZuHwfptzKLICuoAZQonhKT9iFaZ9tylOisYJEif7muCDKJdp3U0gqFN%2B2K0rrNN8A2dlsyLhYFWCxuFTPSgfKSM0KGSzzZd9y%2BZOfHNFgPtU%2B93FSlAeb2Qzask2nsNDlcaqzjHO%2B2hV%2BK6TH7DUogUyxMvqCsGwBjznLlPT9KzyQ8m6pxw9GnxTV1ZsCezB%2BjGrWbNohrx1Wt59yNTITvIP0OHyzgzuDq4m3yjYhnJ4%2FierBvYr%2FaQq77tzQS%2FXMlfWmHafu8C788d%2Fb%2BTj4%2B%2FOZRncemdNDrRRjhgZRqXDhwKgDQ%3D%3D%7Ctkp%3ABk9SR8T848S8Yg
I spray paint the copper side of the board with a dark oil-base paint
(which I have previously tested and found dissolves easily with acetone,
once it needs to be removed), and let it dry.
Then laser scan the track patterns onto the painted surface with the 3018.
I etch with a potent combination of approximately 50ml each of ~30%
hydrogen peroxide and 30% hydrochloric acid and ~50ml of water (reduce the
water for a more aggressive effect).
Wearing protective gloves (of course), I swab the etchant onto the laser
engraved surface of the PCB with a small piece of sponge, and after ~2
minutes, it's done.
Tracks and spaces are very sharp and clean.
Then use acetone to dissolve and wipe off the paint, and the board is done.
Good vid of the etching process at
https://www.youtube.com/watch?v=OTGZcY7WyYI&t.
Hope this is helpful.
Cheers...

Mot007, 2023-08-11:
Hi 104TMR,
Thanks for your feedback.
I would be happy with a 8/8 thou as a minimum. The only way I see right now
is your method or a .2mm very fragile router bit(which is on its way)
The Lazer etching looks impressive!
Involving chemicals is no for me now days, I'm too old and clumsy😂
Be ultra carefull with the chemicals.
That trio (well 2 of them) makes a potential bomb.
Google- Mixing concentrated H2O2 and acetone with an acid catalyst is known
to form the shock and friction sensitive explosives triacetone triperoxide
(TATP) and diacetone diperoxide (DADP)

Cheers
Make sure to keep them chemicals well away from each other! Also in that
video the mix began to runaway and boil, another Hazzlard!

tmr4, 2023-08-13:
I've been having the same height map issues.  Has anyone tried using the average of several height maps?  Did it help at all?

Mot007, 2023-08-14:
Have been working on this and have found some promising results. I'll have to fine tune it and will let you know the fix. 
Here's a quick result. I haven't saved the record before the fix but they where out by about .050mm 
Here's what I got recently
4 probes 30mm square using 60deg V cutter			
probed 4 times different set up. 1 is .001
p1,p2,p3,p4,err
0,	0	0	1	1
13	,12	,12	,14,	2
57	,64,	62,	68,	11
57	,66	,68	,61	,11
				
5	,0,	0,	0,	5
21	,16	,16,	16,	5
57,	55,	50,	49,	8
59,	54	,52,	53	,7

tmr4, 2023-08-14:
I took 5 height map runs for a 40x40 mm board I'm making, discarded the most inconsistent one and averaged the others for the final height map for the board.  The board milled successfully.  The first for this size of board for me.  One sample doesn't prove the technique, but I'm very happy with the results.

I also tried doing the height map of the same area with different types of bits and even the flat end of a bit shaft.  I found that:
- most times the first height map after changing a bit was a throw away and if not one of the others was different enough to consider not using it;
- I occasionally saw a spark as the V bit approached the circuit board.  Could this account for map irregularities?  I can't say as I didn't observe closely enough to make an assessment.  
- the maps made using the flat end bit had the lowest standard deviation of all of the bits.  This makes sense since each probe itself is effectively averaging a very large area around what would be measured by a V bit.

I used an average of the flat end maps for my successful board.

Mot007, 2023-08-15:
Hi Tmr4 
Good for you to getsome +be results! 
What I have found is that the firmware does not denounce the probe input adequately.
But having minimal software expertise, I've resorted to a hardware mod on my driver board. Which is a Chinese derivate using a stm38f030 arm controller. Somewhere along the translation they lost the recipe. I don't believe it is the Candle software at fault but the glue software that goes with the GRBL firmware or the GRBL firmware itself.
I can't pinpoint it but my hardware  mod to my Chinese PCB has allowed me to get a much more precise leveling map that I could ever achieve before. I can live with +/-10 microns on copper probing with a sharp .2mm probe. I take the second reading as the correct one.
All the best.

104TMR, 2023-08-15:
Hi MOT007,
That sounds pretty impressive.  I take it 'denounce' should be
'debounce'(?).
For those who want to pursue routing - and just for the sake of getting
better height maps - are you willing to share the details of the mod you've
made?
I'm sure others would find it helpful.
Cheers...

Mot007, 2023-08-19:
Hi TMR104,
Have put together a quick schematic.
This works for me to an extent but cannot get a 100% fix without understanding what the software is doing.
If I had bought an Arduino type driver board, I could have dived into the code but as this is not Arduino, I don’t even know where to begin to find the source code ( it there is one as it came out of China)
 So for me it is a bandaid fix with better results.
You don’t have to modify your main Driver to try it out. I built the test externally when I was first investigating this issue.
King regards,
Tom

Mot007, 2023-08-19:
[probe1.pdf](https://github.com/Denvi/Candle/files/12384895/probe1.pdf)

104TMR, 2023-08-19:
Hi Tom,
That looks like a very simple, but effective, solution.
Great work.
10 micron reproducibility is pretty good.
Thanks for sharing.
Cheers,
Glen

tmr4, 2023-08-19:
The software (GRBL) is just monitoring the probe pin in probe.c probe_state_monitor().  The probing cycle takes place in  in motion_control.c mc_probe_cycle().  It looks like it is simply checking if the probe pin goes high or low (this is configurable)  during the stepper ISR.  I don't think there is any software debouncing.

Mot007, 2023-08-19:
I'd be happy if someone tried the probe to say if it work for them or not. I don't have much confidence on my Chinese supplied hardware. For instance , I changed $102 to half the value (800 to 400) and had no change on the distance travelled🤔

tmr4, 2023-08-19:
Perhaps I'm misunderstanding your circuit or we have different hardware.  The voltage across my z-probe is greater than 11 volts.

Mot007, 2023-08-19:
Hi TMR4
Firstly, I'm surprised at your probe voltage for today's type logic runs topically either at 5 O 3V. Check your  CRO for X1 or X10 setting.🙂
You seem to have found the part of the proving in the ISR which makes sense as a non software person, correct me if I'm wrong, I presume the stepper is called by a timer ISR as the stepper chip does not provide an Interrupt. Correct?
So I'm assuming the probe does not have its own  IRQ pin and is basically polled. So firstly there is latency in servicing the ISR, then there is the time for a step to occur on the stepper, then there is the chance that the probe input is High when polled (because of bounce), abd there goes another time added to the  number of steps before a probe touch is recognised by the software. So the more noise the deeper into the copper, the more time elapsed the bigger the error. Not having all these timming values makes it hard to pinpoint a fix. Also I may be totally wrong! 😉
Some more software enlightenment is needed to get to the bottom of this!
Cheers 
Tom

Mot007, 2023-08-19:
Try and measure with a multimeter instead of the oscilloscope. What
hardware are you using?
Cheers

tmr4, 2023-08-19:
I get the same probe voltage with a multimeter.  I have the Genmitsu 3018 ProVer V2.  There must be some level conversion going on.  It might be hidden in the [probe](https://www.sainsmart.com/products/photoelectric-z-axis-tool-setting-touch-probe-with-indicators).  I haven't been able to find much information about it.

I think your understanding of the probe routine is correct.  Here is the [probe state monitor](https://github.com/grbl/grbl/blob/master/grbl/probe.c#L56):

```
// Monitors probe pin state and records the system position when detected. Called by the stepper ISR per ISR tick.
// NOTE: This function must be extremely efficient as to not bog down the stepper ISR.
void probe_state_monitor()
{
  if (sys_probe_state == PROBE_ACTIVE) {
    if (probe_get_state()) {
      sys_probe_state = PROBE_OFF;
      memcpy(sys.probe_position, sys.position, sizeof(sys.position));
      bit_true(sys_rt_exec_state, EXEC_MOTION_CANCEL);
    }
  }
}
```

Mot007, 2023-08-20:
TMR4, are you able to tell me what processor your 
Genmitsu 3018 ProVer V2 has and the version it returns via $i

tmr4, 2023-08-21:
Mot007, I haven't seen the specific processor listed, but the specs say it's a 32-bit ARM.  $I returns [VER:ARM32 V2.2.20220826:] [OPT:VZL,35,254]

Mot007, 2023-08-21:
[image]
The top trace is the probe input (trigger low) 
The bottom trace is the Step pin on the Z controller. (Allegro A4988, Chinese version)
As you can see the stepper continues for several pulses after the probe input goes low. I did this many times and the pulse numbers varied. All I can think is that the GRBL software in the ISR ( which sets the call to the probe monitor) is skipping due to who knows what. 
The mapping has improved after removing bounce from the input but I'm still getting wrong values. 
A good Firmware person should be able to debug and fix  this problem.

Mot007, 2023-08-21:
> Mot007, I haven't seen the specific processor listed, but the specs say it's a 32-bit ARM. $I returns [VER:ARM32 V2.2.20220826:] [OPT:VZL,35,254]

You should check your probe wiring, and possibly 12V coning from the motor PS!

tmr4, 2023-08-22:
My z-probe has a three pin connector.  The control board labels for these are 12V, GRD and Pb which I assume stands for probe. At the control board, the voltage of pin 12V is 12.6 volts and pin Pb 11.6 volts, the same as I measured previously between the probe alligator clip and ground plate.  I don't know how Pb is driven; a schematic isn't available.  The control board is in an opaque case.

I get the following if I power the z-probe separately at my workbench with about 12.5V in and Pb pulled up to that through a 1k ohm resistor, just because.  The probe can't be disassembled for inspection.

[image]

The yellow trace is the signal I've shown before, between the z-probe alligator clip and its grounding plate.  Here I've shown the alligator clip touching the grounding plate for about 280 ms.  The magenta trace is pin Pb relative to GND.  

The rise of the magenta trace looks similar to the bounce damping I was seeing before.  If the alligator clip is left touching the grounding plate, the magenta trace rises to 0 volts after about a half second.  This reverses when the alligator clip is raised.  

I could not cause a bounce during these tests so I'm not sure how the magenta trace responds with that.

I want to look at the z stepper signal as well.  It seems like a cause for the variability you're seeing.  Unfortunately, I won't be able to work on this for a couple of weeks.

Mot007, 2023-08-23:
Here's more food for thought. From Google.

Usually, the surface of the conductor in printed circuit boards is intentionally roughened to enhance the adhesion to the prepregs. Typical surface roughness Rz of the copper foil commonly used in printed circuit board is 6 μm
🤔

poes8943, 2024-02-11:
copied from #626 
test steps using Z-probe control button:
    push Z-probe control button and perform a Z-zero after that (work coordinates now at 0.000 0.000 0.000)
    jog Z-up 0.5mm and pushing Z-probe control button again (Z axis work coordinate at 0.399 now)
    jog Z-up 0.5mm and pushing Z-probe control button again (Z axis work coordinate at 0.807 now)
    jog Z-up 0.5mm and pushing Z-probe control button again (Z axis work coordinate at 1.199 now)
    ... there seems to be roughly an error of 0.4mm after each step and values adding up
    ... looks to me like there is one additional full turn of the spindle(?) (4mm)

same test with manual commands:
    push Z-probe control button and perform a Z-zero after that (work coordinates now at 0.000 0.000 0.000)
    jog Z-up 0.5mm and executing 'G38.2Z-2F10' (Z axis work coordinate at 0.009 now)
    jog Z-up 0.5mm and executing 'G38.2Z-2F10' (Z axis work coordinate at 0.006 now)
    jog Z-up 0.5mm and executing 'G38.2Z-2F10' (Z axis work coordinate at 0.007 now)

Hostaler, 2024-02-11:
> Hi Denvi, Many thanks for a great piece of software. I'm using Candle 1.1.7 with a Vevor 3018 CNC mill to mill PCBs.
>
> I undertake a height map before any PCB mill, and tick the box in Candle to use [...]

 How do I apply all this so as not to lose precision?

Hostaler, 2024-02-11:
How do I apply all this so as not to lose precision?

Hostaler, 2024-02-11:
How do I do all this to regain precision in my grbl? In which section
should I put it?