Denvi/Candle #554: Gcode in probe and command buttons doesnt stay after restart

Amrt0623, 2022-05-10:
Is your feature request related to a problem? Please describe.**
After restarting candle i always have to reenter probe and command button gcode

**Describe the solution you'd like**
Some way to save the current gcodes

ondras12345, 2022-05-10:
These options should be stored in `settings.ini` (probe command is `touchCommand=`).
Try changing some other settings (e.g. visualizer colors). Is the change persistent after restarting Candle?
Which version of Candle are you using? Which operating system?
Do you have write access to the `settings.ini` file?

Amrt0623, 2022-05-10:
Newbie not a clue. Where do i find settings.ini and touchCommand=
Colours also revert back to orignal.
Version 1.1.8
Mac os high sierra.  Ver10.13.6
No access to settings.ini

ondras12345, 2022-05-11:
Candle needs to have write access to the directory where `Candle.exe` is, because it needs to create a file called `settings.ini` there.

mar0x, 2022-05-11:
Hello @Amrt0623 .
- Make sure you have installed Candle to **Applications** (or any other place) and start an _installed_ application (not from opened .dmg archive).
- Was there any application crash messages when you quit from Candle?

Amrt0623, 2022-05-11:
I am certain i installed all my application.  But i will repeat the process

no crash messages appeared

Amrt0623, 2022-05-11:
Also i seem to have loose my mm/turn calibration as well

Amrt0623, 2022-05-11:
Candle app is in the download folder. I have not moved it.  Candle app appears have over written previous last file.  However date modified remains. Jan 3 2020 at 16:54. This predated my first download.   Opened changed colour of tool closed reopened.  Tool reverts to original color

Amrt0623, 2022-05-11:
Cant find a settings.ini file anywhere???

mar0x, 2022-05-11:
CH34x_Install_V1.4.pkg - according to the name, it's a driver for USB-serial chip. It maybe required for your board, but not related to Candle application.

settings.ini file should be inside Candle.app directory (Candle.app/Contents/MacOS/settings.ini).

Please calm down and carefully describe your environment and issue(s).
- what is the CNC you are using (model, name, board, URL etc)?
- what is the GRBL firmware version on it?
- where did you get your Candle.app?
- why do you put it in Downloads instead of Applications?

Amrt0623, 2022-05-11:
Genmitsu proverxl  
Grbl 1.1 compatible 
Aurduino board cant more data
Download from sainsmart.  ( could re download installer)
Installer gives option but says don’t change default locations. Thats where it went.

Issue is that no changes i make in preferences return after closing and then restarting candle

mar0x, 2022-05-11:
Please share the link to installer for investigation or use "unofficial" [Candle-1.1.9.dmg](https://github.com/mar0x/Candle/releases/download/v1.1.9/Candle-1.1.9.dmg) - open it and copy Candle.app to your favourite location.

Amrt0623, 2022-05-11:
K tried to open candle app.  Wont open just runs.  So no access to settings.ini

will try candle-1.1.9.dmg

mar0x, 2022-05-11:
I've read [the instructions](https://docs.sainsmart.com/article/zcyhab7a0k-how-to-install-grblcontrol-candle-for-mac) and after candle.dmg open should be Candle.app **copy** to Applications or any other place.

Amrt0623, 2022-05-11:
https://docs.sainsmart.com/article/vdgvaufcu1-genmitsu-prover-xl-4030-resources

Scroll down

Amrt0623, 2022-05-11:
Found settings.ini.    Aaah

Amrt0623, 2022-05-11:
So saved an original copy and will modify settings.ini.   With my updates.  Hooe it gies well.   Thank you very much

Amrt0623, 2022-05-11:
K copied gcodes userCommands0,1,2,3
G code is approximently the same length as z probe command.  
Z probe command change is retained

But when i open candle.  I only get the first two codes in the buttons G21G91 so no action when pressed

Also changed port to tty.wchusbderial430. But it was not retained either?

Amrt0623, 2022-05-11:
Got it added quotes. Wow didnt think i was ever gonna get this. So much thx

closed 2022-05-11 by Amrt0623

ondras12345, 2022-05-11:
You shouldn't need to edit `settings.ini` by hand. If the permission in the filesystem are set correctly, Candle should be able to modify it automatically.

mar0x, 2022-05-11:
The thing is the information about `settings.ini` and permissions is too technical and it was provided too early, prior the issue analysis. The real issue was (I suppose) in incorrect Candle.app installation (the absence of installation, to be precise).