Denvi/Candle #557: Candle 1.2b jog bad number format

PaulHosh, 2022-06-09:
Hi Denvi
thanks for the candle code.

A while ago, I used the Candle on a 3018 machine with success. Now I have built a new CNC with UNO. Using GRBL commands in the serial monitor (Arduino IDE) I can move the spindle as requested (up down, left right) with no problems. 

I had a NC file made by fusion 360 which I loaded and surprise! it moves the spindle as expected. but I cannot jog the spindle with either Candle 1.1.7 nor Candle 1.2.13b. With Candle 1.2.13b I get a bad number format error.  I looked at the GRBL documentation and the command string seems ok

I would like suggestions to get the jog feature to work. 
thanks
kind regards
paul

[image]

mar0x, 2022-06-09:
Hello. According to prompt, your machine has Grbl 0.9j. Candle 1.1 and 1.2 requires Grbl 1.1 for jog control.

mar0x, 2022-06-09:
Although I cannot see attached images, I agree the 1.2 version has many useful features. To be able to use it with your CNC machine, you need to upgrade GRBL firmware up to 1.1 version.

PaulHosh, 2022-06-10:
HI Max

Thanks for your help.

When compiling GRBL 1.1h, I have encountered the errors noted at
https://github.com/gnea/grbl/issues/390.

Required HOMING_CYCLE_0 not defined.
WCO refresh must be greater than one.
Override refresh must be greater than zero.

Further reading told me to remove all previous copies of GRBL from the
Arduino libraries folder. This I did. I did a search of the entire C: drive
and removed anything remotely looking like GRBL. I then downloaded GRBL
1.1h. Following the instructions (right/wrong?) at
https://www.youtube.com/watch?v=AeuFev3Bv74 I copied the GRBL folder to the
Arduino libraries folder.

Then this is where I am not quite sure what should happen. I opened the
Arduino IDE, selected New (file) and chose GRBL from the Include Sketch
menu.  I connected a UNO to the USB and the port was found. I then compiled
and encountered the compile errors.  I did not make any changes to any
file.

The discussion for issue 390 says to comment out some lines, uncomment
others for the Homing cycle error.
And there is a note about changing (or not ) other variables to resolve the
second and third compile error.

My CNC machine I built using standard NEMA 23 motors, drive controllers and
a UNO.  So I think that the comment about the CoreXY does not apply.

When looking for the GRBL files to remove, I found a zip file for version
1.1h that I got two years ago.  I tried that and had the same compile
errors.

Thank you for your patience. Please advise how I should proceed.
Kind regards
paul

mar0x, 2022-06-10:
Hi Paul,

Please read carefully the official [Compiling Grbl](https://github.com/gnea/grbl/wiki/Compiling-Grbl) article. **Just** follow the instructions (no "_Copy-And-Paste_", no "_New (File)_", no need to choose "grbl" from "Include.." menu etc.). I'm sure you'll compile it!

PaulHosh, 2022-06-11:
Thanks Max

PaulHosh, 2022-10-11:
Hi Max

success!!   I had two versions of Arduino IDE 1.8.15 and 1.8.16. The 1.8.15
compiles but the other did not.

so I downloaded IDE 1.8.19.  ALL is Good!!  I now can jog the machine.

thanks!!

mar0x, 2022-10-11:
Hi Paul,

I'm glad you did it!

Artem123123123, 2025-02-18:
[image]
I just make new buttons that work fine
Service -> User commands

closed 2025-09-08 by github-actions[bot]