Denvi/Candle #565: Bug: Personalizing the interface Candle 1.2.15b

ArturoPin, 2022-07-03:
**Versions**
 - Candle version: Candle 1.2.15b
 - Firmware version: 1.1h

**PC info**
- OS: Windows 10 Home Single Language, Version: 21H2
- OpenGL version: version: 4.6

**Describe the bug**
I would like to personalize my Candle 1.2.15b interface, that's included on this version. But the changes are not saved when close. (when I open candle again It is as default.)

How I would like to have it:
[image]

When I open it again:
[image]

mar0x, 2022-07-04:
@ArturoPin please describe how did you installed the Candle, how do you usually start it and what happens when you close it (maybe there is a message about application crash?)

mar0x, 2022-07-04:
@ondras12345 What is the **typical** sequence of Candle installation which should lead to absence of write permissions?

ondras12345, 2022-07-05:
> What is the **typical** sequence of Candle installation which should lead to write permissions absence?

Unzip as administrator to Program Files and try to run as a regular user. I haven't used Windows for a few years now, but i vaguely remember something like that happened to me.

mar0x, 2022-07-05:
@ondras12345 "Unzip as Administrator" is not the **typical** case for Windows 10 Home, don't you think?

"Running _directly_ from ZIP" - is what I'm expecting to get as an answer. Permissions is not an issue and the suggestion to check the permissions is confusing.

rjcamatos, 2022-07-05:
Extracting to Programa Files folder it Will adquire the permisions of that
folder, by default no site to users.

ArturoPin, 2022-07-05:
You were right, all what I need to do was running the SW as administrator. Thanks

closed 2022-07-05 by ArturoPin