Denvi/Candle #566: Candle just stop send G code lines

vilian12306, 2022-07-03:
**Versions**
 - Candle version: 1.2.15b
 - Firmware version: 

**PC info**
- OS: …
- OpenGL version: ...

**Describe the bug**
When in Run state,I open a CNC G code file and touch Send command.Candle just send a few lines then stop.
Attainments are G code file,Bus Hund traced files and Candle UI picture.
So I want to know in what condition does Candle stop sending?

**To Reproduce**
Steps to reproduce the behavior:
[Candle Test.zip](https://github.com/Denvi/Candle/files/9034733/Candle.Test.zip)

1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Example files**
Add G-code files that cause problems.

**Additional context**
Add any other context about the problem here.

mar0x, 2022-07-05:
Hello.
Please enable program gcode output to console, reproduce the issue and share console output and your GRBL settings (response for '$$' command). The screenshot of the full Candle window after stop would be helpful too.

It is possible your machine reach the limit (hard or soft).

Playerarrs70, 2022-10-22:
same with me here. the program stops working and the servo spins insane. Tried to reset it and start over again but ended in exact line gcode and exact problem.

mar0x, 2022-10-22:
@Playerarrs70, same for you [up there](https://github.com/Denvi/Candle/issues/566#issuecomment-1174716969).

Playerarrs70, 2022-10-24:
@mar0x problem fix sir, thanks for the reply. Looks like I've been making mistake with calibration and my arduino damaged because of rust. So it's hardware and human error. My bad 😔

Playerarrs70, 2022-10-29:
Here some screenshots. 
[image]

As you can see the program stop when I send input to draw the horizontal line. It doesn't happen in vertical line, at least yet. And last week (when my first comment) it's worse.

[image]

As you can see in the screenshot it stop on the red circle on T letter. I thought it was calibration error so I shrink the $100,101,102 value and it works. But it's happen again with longer line

And here's my settings
Grbl 1.1h ['$' for help]
$$ < $0=10
$1-25
$2=0
$3=0 $4-0
$5-0
$6=0
$10=1
$11-0.010
$12=0.002 
$13-0
$20=0
$21-0
$22-0
$23-0
$24-25.000
$25=500.000 
$26=250
$27=1.000
$30-1000 $31-0
$32=0
$100-781.250
$101-781.250
$102-78.000 
$110-5000.000
$111-5000.000
$112-5000.000
$120-200.000
$121=10.000
$122-10.000 $130-200.000
$131-200.000 $132-200.000

mar0x, 2022-10-29:
According to console output on the first screenshot, CNC report "error:22" which is caused by [invalid feed rate](https://github.com/gnea/grbl/blob/bfb67f0c7963fe3ce4aaf8a97f9009ea5a8db36e/grbl/report.h#L45)  (F0.0). Please fix your gcode program.

mar0x, 2022-10-29:
Consider use "Check mode" (see the check box at the left bottom) before actual run.
What software you are using to generate the gcode? Perhaps it requires configuration to generate proper gcode.

Playerarrs70, 2022-10-30:
Thanks for the reply and advice. I use this app called StickFont here's some screenshots of the app. This app can convert text to gcode instantly. I have seen some tutorial from the internet to convert text to gcode and I found this app. Any advice for app to convert text to gcode ? 

[image]

Lastly may i ask for proper nc file for just test ? I haven't make my machine running smoothly without error.

mar0x, 2022-10-30:
It seems you need to adjust "Plunge feedrate" and "Feedrate" values. It's a spindle move speed in "mm/minute". Start from "50" and increase if you think your machine may mill faster.

I usually use Inkscape with Gcodetools. The user interface is not user-friendly and it requires some experience. It's a good idea to review (and fix) produced gcode files in text editor. See attached [system_text.zip](https://github.com/Denvi/Candle/files/9895761/system_text.zip) with .svg and produced .ngc files.

closed 2025-09-07 by github-actions[bot]