Denvi/Candle #569: Automatic Leveling

Franco-MB, 2022-07-31:
Hi, I apologize for my English!
After a week of incessant searching here I am begging for help…
After running the Auto Leveling I noticed that the survey data is not being identified. The CNC travels as expected according to the probe grid number.

[image]

I noticed that the same happens with the controller (bcnc).
I can operate normally with the CNC.

**My GRBL settings**
$0=10
$1=25
$2=0
$3=6
$4=0
$5=0
$6=0
$10=1
$11=0.010
$12=0.002
$13=1
$20=0
$21=1
$22=1
$23=0
$24=25.000
$25=800.000
$26=25
$27=1.000
$30=1000
$31=0
$32=0
$100=400.000
$101=400.000
$102=400.000
$110=800.000
$111=800.000
$112=500.000
$120=500.000
$121=500.000
$122=500.000
$130=180.000
$131=300.000
$132=45.000

Franco-MB, 2022-07-31:
After a new search I could see problem #563, I will proceed with the upgrade to GRBL 1.1h. I apologize for the little knowledge and thank you for all the effort in solving the problems. I will let you know if it works!

Franco-MB, 2022-07-31:
Updated to v1.1h (2019-08-25) Release, everything working as expected.

ondras12345, 2022-08-04:
I'm glad you solved your problem.
Please close this issue if you have no further questions.

Franco-MB, 2022-08-12:
So far everything is working!

closed 2022-08-12 by Franco-MB