Denvi/Candle #570: Load program using barcode scanner

gozeen, 2022-08-04:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

gozeen, 2022-08-06:
The exact requirement is, a barcode is labeled on the workpiece, which when scanned it loads the specific file for that workpiece out of many files created and saved. 

Ok, here's the problem i have to machine about 500-600 pieces per day which almost look the same, meaning about 500 different g-code files that has to be selected before machining, this increase the chance of errors by selecting wrong code, and also to save time, this system would streamline my production and make sure anyone can machine it error free

ondras12345, 2022-08-06:
Alright, how about this:
- Most hardware barcode scanners behave like keyboards. They enter the characters found in the code and then press return. (I believe this is configurable using special barcodes found in the manual.)
- Ctrl+O brings up the Open file dialog in Candle. Depending on your operating system, it might be possible to just start typing a filename and press return to open that file.
- The dialog seems to always start in the directory the last program was opened from.
- If you put all your Gcode files in the same directory with reasonably short names (so that the barcodes can be easily read), you should be able to just hit Ctrl+O and scan a code with e.g. `0001.nc` in it. That should open the file in Candle.

gozeen, 2022-08-08:
OMG that sounds amazing let me try that, well first i would have to buy a barcode scanner then try it, also a barcode generator,  also a printer, 

Then the files have to be properly named and barcodes to be printed accordingly. 

I was thinking of adding a switch to the scanner holder, so when i lift the scanner to scan it triggers and presses Ctrl+O, then i scan it. How dose that sound.

gozeen, 2022-08-08:
Well, i am not a programmer, i am mechanical engineer, but into mechatronics and automation, 

So writing a script would be something new but i know how to kinda, 

No there are no barcodes on the workpiece, we run a panel processing unit/ modular furniture manufacturing plant, and i  trying to fix the labor issue, by automating part of the production. 

I could just invest on a 6axis drilling cnc, but thats not gonna be possible due to resent losses we faced.

Anyway thanks for showing interest,  in my little endeavor, its been fun

closed 2025-09-07 by github-actions[bot]