Denvi/Candle #571: Heightmap Probe Always Starts at X0Y0 Regardless of Boundary Limits

frederickbrown, 2022-08-06:
**Versions**
 - Candle version: 1.1.7
 - Firmware version: 1.1h

**PC info**
- OS: win7
- OpenGL version: 3

**Describe the bug**
Candle always starts probing the height map at x0 y0 regardless of what I said for the map boundary. I am Milling pcbs, and this causes a problem because x0 y0 is the edge of my pcb. The probe usually slips down along the corner of the PCB without triggering a z value. I am only interested in mapping the area within the boundary I set, which is somewhat smaller than the Raw pcb. I have tried to compensate by manually triggering the Z detection with an alligator clip when the Milling probe reaches my previously set z0. I have found this leads to inaccurate Milling depths on the PCB afterwards.

Maybe starting the height map at x0 y0 then moving to map the boundary area is the intended Behavior of the program, although for me it just creates a problem.

**To Reproduce**
Steps to reproduce the behavior:
1. Set X0Y0
2. Click create height map.
3. Set boundary starting somewhere inward of X0Y0.
4. Click probe.
5. Watch candle start the height map probing at C0Y0 the move to the start of the boundary area and map that.

**Expected behavior**
I expect the height map probing to start at the lowest X and lowest y point in the boundary area.

ondras12345, 2022-08-06:
Duplicate of https://github.com/Denvi/Candle/issues/544

mar0x, 2022-08-06:
(0, 0) is a "reference point" for heightmap. Possible options to solve this and similar issues are:
- use latest Z probe touch height as a reference;
- configure reference point coordinates at the heightmap edit screen;
- combined solution: coordinates and latest Z probe height are displayed at the heightmap edit screed and can be edit;

frederickbrown, 2022-08-07:
For some reason when I hit the z probe button, my machine just stops and becomes unresponsive. I then have to reset, unlock and set z0, y0 and z0 again.

Anyway, I set up my CNC files to set the center of the board at x0 y0. That works with this project. Although it would be ever so helpful, if you could set up the program so that the height map reference point position is user settable. I see users have been asking for this solution literally for years now. I'm sure some thing can be done.

Thank you for the quick responses.

DEF7, 2022-11-28:
> For some reason when I hit the z probe button, my machine just stops and becomes unresponsive.

This sounds like the probe state is already triggered when you hit Z-probe. In other words, the controller things it's already touching/contacting something. This happens to me when I'm heightmapping if I don't set the upper Z limit for the probing high enough and a high spot on the workpiece causes the probe to crash into it and then when it tries to probe it errors out with an alarm.

It sounds like you need to change your GRBL's probe pin to be the opposite. Is your probe an always-on style or is it supposed to complete a circuit upon contact? The GRBL setting is the $6 probe pin invert one. See here: https://github.com/gnea/grbl/blob/master/doc/markdown/settings.md#6----probe-pin-invert-boolean

I have the invert probe pin setting set to '1' (i.e. $6=1) because I use an always-on style probe. It was really frustrating when my touch probe became unreliable though. It would just start randomly thinking it was contacting when it wasn't due to corrosion on some of the crummy connections between the internal contacts, and it would cause the same error situation. I too would have to reset my machine zero when this happened. However, I found out that if I click Reset more than once, before unlocking and resetting the machine, it seems to restore my machine zero and I don't have to set it again! This was a huge relief because it was very taxing when my probe started acting up. Clicking reset once 

Building a new probe from scratch right now and resolving the issues the original one had:
[image]

closed 2025-09-07 by github-actions[bot]