Denvi/Candle #573: 1.2.14b source code

vvv1954, 2022-08-28:
Could you upload please source code of 1.2.14b version?

Thanks in advance.

ondras12345, 2022-08-29:
Check out the Experimental branch.
https://github.com/Denvi/Candle/tree/Experimental

However, the beta releases aren't tagged, so it's hard to guess which commits correspond to them.

vvv1954, 2022-08-30:
Thanks. User Commands disappeared?

ondras12345, 2022-08-30:
I believe they are available as a built-in plugin.

XRay3D, 2022-11-24:
I added the assembly through CMakeLists, it build faster and easier. QMake is dying. 
https://github.com/XRay3D/Candle/tree/Experimental
Denis does not really want to update the Qt version for some reason.

Perlovka, 2022-11-26:
@XRay3D good try, but now it depends on some third-party tool (cqtdeployer) to compile.

XRay3D, 2022-11-28:
> cqtdeployer

cqtdeployer собирает в папку программ библиотеки, переводы и ресурсы Qt сам. Вы можете не использовать его, но тогда все это нужно переносить вручную.

closed 2025-09-07 by github-actions[bot]