Denvi/Candle #577: Z probe/heightmap stops and alarms with error 9

alxkos, 2022-10-10:
**Versions**
 - Candle version: 1.17, 1.2.15b
 - Firmware version: 1.1h

**PC info**
- OS: windows
- OpenGL version: 
**Describe the bug**
Having a problem with both z probe and heightmap, I have 3018cnc, using simple two crocodiles probe connected to the board. 
Z probe command is 
G21G91
G38.2Z-30F100
G0Z2
G38.2Z-2F10
Once tool touches the pcb it stops and i'm getting error 9 in the console and it doesnt continue. stops on the first step. 
Im sure heightmap not working for the same reason. 
Did I missed something in the setup? 
Thanks

mar0x, 2022-10-11:
Hello,

> Also motor is stopped, but it continue to count Z axis until it gets to -30, then alarms

The motor is stopped by CNC or stopped because it is unable to move lower, keep pushing on the PCB, skipping steps and sounds like 'G-R-R-R-R' ?

I would suggest to check the probe: let your "simple two crocodiles" touch each other, you should see "P" (probe) letter somewhere at the left-top corner of your 3D view.

DEF7, 2022-11-28:
ALARM 5 means that it went the full probing travel that is specified and didn't hit anything. Either the alligator clips aren't forming a complete circuit upon contact or your probing Z-travel is too short for the height you're initiating it at.

e.g. if you are 0.5" above the workpiece or surface you're probing and your Z travel is set to 0.1" for probing, it will travel -0.1Z and then error out having not touched anything.

If it's coming into contact with the PCB then it's probably not forming a complete circuit for the controller to detect. A good test is to initiate a Z-probe with the alligators not connected, and just touch them together yourself by hand, one clip to the other, to see if the Z-probe function is actually working.

If that doesn't work then you have something hooked up wrong. If it does work then there's a conductivity issue happening with what you're clipping them to - either the tool or PCB is not conducting.

I did notice reliability issues when probing with the probe clips on my Arduino/GRBL machine if I put the positive on one and negative on the other. Swapping them fixed it. I don't remember if it was better to put the positive or the negative on the tool, but one was definitely better.

mar0x, 2022-11-29:
The sad thing is we never know what exactly caused the @alxkos 's issue as well as whether it was _fixed_ and _how_ it was fixed.

ousiafb, 2024-07-05:
> The sad thing is we never know what exactly caused the @alxkos 's issue as well as whether it was _fixed_ and _how_ it was fixed.

Hi, @mar0x solved a similar issue for someone today. What worked in our case was just inverting the probe pin with $6 with Grbl 1.1