Denvi/Candle #578: Different machines.

GrahamJB, 2022-10-15:
Is there any chance of a Machine setting being added?

I have a number of routers and each Z-Probe base is a different thickness and they have different Z axis movements, different spindle speeds, movement rates and acceleration values.
To reconfigure Candle each time I use a different router is a real pain, not really viable.
But if I could set up and then select from a list of machines, using the Machine information and Control values for each machine it would make Candle usable.

ondras12345, 2022-10-15:
Workaround: you could copy the directory where Candle is installed and have multiple instances, each one with it's own settings.

GrahamJB, 2022-10-16:
Good idea, I hadn't thought of that, I might use a batch file to copy a settings.ini file in, before starting the program, and prompt to copy it back when candle closes. That may be easier to track and take less space on the drive.

ondras12345, 2022-10-16:
Or even use a symlink to make `settings.ini` point to a machine-specific config file. That way, you don't even need to wait for Candle to close

GrahamJB, 2022-10-16:
That is also a possibility, but I'm on Windows 10 so I think I would have to run as an administrator. I can't upgrade to Windows 11 as my PC does not have a TPM.

closed 2025-09-07 by github-actions[bot]