Denvi/Candle #579: When removing the height map, the machine stops

RuslanLLLLL, 2022-10-16:
Hello everyone!
Error unknown to science!

**Versions**
 - Candle version: , 1.1.7, 
 - Firmware version: 1.1

Windows10

When removing the height map, the machine stops each time at the last point of the first pass along the X axis. The result is repeated over and over again. All other machine functions work.
I reinstalled the Candle program, the Arduino firmware too. Nothing changed.

Help me please

[image]
[image]
[image]
[image]

[image]

mar0x, 2022-10-16:
Hello,

Please describe sequence of operation step-by-step, share your gcode program and produced heightmap (if any).

It would be good to have an information regarding "error: 9" you probably can see at the issue description as well.

mar0x, 2022-10-17:
According to the screenshot 2:
"ALARM: 2" - most likely triggered from [grbl/protocol.c:224](https://github.com/gnea/grbl/blob/bfb67f0c7963fe3ce4aaf8a97f9009ea5a8db36e/grbl/protocol.c#L224) which is in turn caused by [grbl/limits.c:426](https://github.com/gnea/grbl/blob/bfb67f0c7963fe3ce4aaf8a97f9009ea5a8db36e/grbl/limits.c#L426).

[Soft limit](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#20---soft-limits-boolean) is reached and triggered this alarm. I would suggest to check the limit settings.

This is definitely NOT the [grbl issue](https://github.com/gnea/grbl/issues/1169).

mar0x, 2022-10-17:
It is hard to guess what's going on without complete information about coordinates, send commands and received responses.

Try to record the communication between Candle and CNC serial port using serial port monitor.

RuslanLLLLL, 2022-10-18:
Hi,
I did some work here...
Look please.
step by step
[image]
[image]
[image]
[image]

mar0x, 2022-10-18:
Could you please provide serial port communication dump file instead of comics.

RuslanLLLLL, 2022-10-28:
The problem is solved by changing the parameter $20=1 to $20=0.
I wish you all success!

closed 2023-06-01 by RuslanLLLLL