Denvi/Candle #580: working on plugin, how do I use console to post some data?

alxkos, 2022-10-19:
hello, since there is no doc on plugins, i'm digging in the source code, but cannot get how do I send testing data to console?
Thanks

Denvi, 2022-10-20:
Hi, there is some docs here: https://github.com/Denvi/Candle/tree/Experimental/doc
You can view debug output with "DebugView" program.
You can use "print()" function in .js script file to output some data.

[image]

alxkos, 2022-10-21:
> Hi, there is some docs here: https://github.com/Denvi/Candle/tree/Experimental/doc You can view debug output with "DebugView" program. You can use "print()" function in .js script file to output some data.
> 
> [image]

thanks!!!!!!!! checking on

davi-domo, 2024-01-27:
Hi,
I'm a tinkerer, I like candle and adding plugins is great
recently I made a "CANDLE-BLE" bluetooth remote control which sends keyboard shortcuts to candle via bluetooth https://youtu.be/rjfkb--jz7Q

for another cnc milling cutter selection project I am starting a plugin, based on the emergency stop button, which captures the *.nc file (generated by KRABZCAM), extracts the data in comments, before sending it to candle
so far no problem

but I am blocked from sending the HTTP request in GET method to my tool distributor

could you direct me to the qt.network extension?

```
function onButtonClicked() {
    print("my command");
    var fichier = QFileDialog.getOpenFileName(this, ("Open G-CODE"), "", ("G-code files (*.nc)"));
    var fs = new QFile(fichier);
    var i = 0;
    if (fs.open(QIODevice.ReadOnly)) {
        var txt = new QTextStream(fs);

        while (!txt.atEnd()) {
            var textin = txt.readLine()
            if (textin.indexOf("BEGIN TOOLPATH ") != -1) {
                const indexOfFirst = textin.indexOf("'");
                const indexlast = textin.indexOf("'", indexOfFirst + 1);
                var result = textin.substring(indexOfFirst + 1, indexlast);
                print("YESSSSSSSSSSSSSSSSSSSSSSSSSSS ");
                print(result);
                var url = new QUrl('192.168.0.36/test?t=' + result);
                var send = new QNetworkRequest(url);
                print(send);
                break;
            }
        }
        app.loadFile(fichier);
    }
}
```
I'm not asking for a code but for information to understand how it works
after a sleepless night I'm a little desperate 
thanks

davi-domo, 2024-01-27:
Hi,

I finally succeeded, I share it can help
I don't know if it's the best way but it works 😉
```
function onButtonClicked() {
    print("my command");
    print("");

    var fichier = QFileDialog.getOpenFileName(this, ("Open G-CODE"), "", ("G-code files (*.nc)"));
    var fs = new QFile(fichier);
    var i = 0;
    if (fs.open(QIODevice.ReadOnly)) {
        var txt = new QTextStream(fs);

        while (!txt.atEnd()) {
            var textin = txt.readLine()
            if (textin.indexOf("BEGIN TOOLPATH ") != -1) {
                const indexOfFirst = textin.indexOf("'");
                const indexlast = textin.indexOf("'", indexOfFirst + 1);
                var result = textin.substring(indexOfFirst + 1, indexlast);
                print("YESSSSSSSSSSSSSSSSSSSSSSSSSSS ");
                print(result);
                var url = new QUrl('http://192.168.0.36/?t=' + result);
                var manager = new QNetworkAccessManager(app);
                var req = new QNetworkRequest(url);
                var send = manager.get(req);
                break;
            }
        }
        app.loadFile(fichier);
    }
}

```
[image]

[image]

[image]

Hostaler, 2024-01-31:
Hola buenas tardes davi-domo, serias tan amable de decir de donde se puede bajar tu nueva versión de Candle?.
         Muchas gracias por adelantado.

closed 2025-09-07 by github-actions[bot]