Denvi/Candle #583: Tool not showing.

jeffknodle, 2022-11-01:
When I start machining a part the animation of the tool only shows rotation. It does not follow the tool path. It stays on the start of the path and just "spins". Am I missing a setting somewhere? I'm using a Sainsmart 3018 Prover.

ondras12345, 2022-11-01:
Please post your [Grbl settings](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration) (`$$` in console).
Do the Machine coordinates / Work coordinates in the top right corner change when the machine moves?

jeffknodle, 2022-11-02:
> Please post your [Grbl settings](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration) (`$$` in console). Do the Machine coordinates / Work coordinates in the top right corner change when the machine moves?

No the coordinates do not change when the machine is moving.
I'm new to this forum and using my 3018, so any help would be welcome.

jeffknodle, 2022-11-02:
Update: I installed grbl candle version 0.8 and it works. I'm assuming there is a bug with version 1.1.7.

ondras12345, 2022-11-02:
Candle != Grbl

Look at the Console panel in Candle. What does it say when the machine starts? It should print something like `Grbl 1.1h ['$' for help]`.
Type `$$` to get the values of all Grbl settings and post them here.
This is likely to be an incompatibility between your version of Grbl and Candle 1.1.7 (which only supports Grbl 1.1) rather than a bug.
If you are running Grbl 1.1, try setting `$10=1`.

mar0x, 2022-11-02:
Hello,
@jeffknodle I would suggest to follow the [bug report template](https://github.com/Denvi/Candle/blob/master/.github/ISSUE_TEMPLATE/bug_report.md) and provide comprehensive information about your setup. This dramatically increase chance to get proper advise or workaround the issue you face.

MPCMousavi, 2023-07-05:
hello friends
I also encountered this problem, has anyone been able to solve this problem?

MPCMousavi, 2023-07-05:
> شمع != گربل
> 
> به پنل کنسول در Candle نگاه کنید. هنگام روشن شدن دستگاه چه می گوید؟ باید چیزی شبیه به `Grbl 1.1h ['$' for help]`. برای دریافت مقادیر تمام تنظیمات Grbl تایپ کنید و آنها را در اینجا ارسال کنید. این احتمالاً یک ناسازگاری بین نسخه Grbl شما و Candle 1.1.7 (که فقط Grbl 1.1 را پشتیبانی می کند) است نه یک اشکال. اگر از Grbl 1.1 استفاده می کنید، تنظیمات را امتحان کنید .`$$``$10=1`

Hello
I also found the same problem by changing the value of 2 to 1 in $10, the problem was solved.
Thank you for your help, dear friend

MPCMousavi, 2023-07-05:
> Candle != Grbl
>
> Look at the Console panel in Candle. What does it say when the machine starts? It should print something like `Grbl 1.1h ['$' for help]`. Type `$$` to get the values of all Grbl [...]

Hello
I also found the same problem by changing the value of 2 to 1 in $10, the problem was solved.
Thank you for your help, dear friend

MPCMousavi, 2023-07-05:
<html>
<body>
<h4 dir="auto" style="box-sizing: border-box; margin-top: 24px; margin-bottom: 16px; font-size: 16px; font-weight: var(--base-text-weight-semibold, 600); line-height: 1.25; color: rgb(31, 35, 40); font-family: -apple-system, BlinkMacSystemFont, &quot;Segoe UI&quot;, &quot;Noto Sans&quot;, Helvetica, Arial, sans-serif, &quot;Apple Color Emoji&quot;, &quot;Segoe UI Emoji&quot;; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; white-space: normal; background-color: rgb(255, 255, 255); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;">$10 - Status report, mask</h4><p dir="auto" style="box-sizing: border-box; margin-top: 0px; margin-bottom: 16px; color: rgb(31, 35, 40); font-family: -apple-system, BlinkMacSystemFont, &quot;Segoe UI&quot;, &quot;Noto Sans&quot;, Helvetica, Arial, sans-serif, &quot;Apple Color Emoji&quot;, &quot;Segoe UI Emoji&quot;; font-size: 16px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; white-space: normal; background-color: rgb(255, 255, 255); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;">This setting determines what Grbl real-time data it reports back to the user when a '?' status report is sent. This data includes current run state, real-time position, real-time feed rate, pin states, current override values, buffer states, and the g-code line number currently executing (if enabled through compile-time options).</p><p dir="auto" style="box-sizing: border-box; margin-top: 0px; margin-bottom: 16px; color: rgb(31, 35, 40); font-family: -apple-system, BlinkMacSystemFont, &quot;Segoe UI&quot;, &quot;Noto Sans&quot;, Helvetica, Arial, sans-serif, &quot;Apple Color Emoji&quot;, &quot;Segoe UI Emoji&quot;; font-size: 16px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; white-space: normal; background-color: rgb(255, 255, 255); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;">By default, the new report implementation in Grbl v1.1+ will include just about everything in the standard status report. A lot of the data is hidden and will appear only if it changes. This increases efficiency dramatically over of the old report style and allows you to get faster updates and still get more data about your machine. The interface documentation outlines how it works and most of it applies only to GUI developers or the curious.</p><p dir="auto" style="box-sizing: border-box; margin-top: 0px; margin-bottom: 16px; color: rgb(31, 35, 40); font-family: -apple-system, BlinkMacSystemFont, &quot;Segoe UI&quot;, &quot;Noto Sans&quot;, Helvetica, Arial, sans-serif, &quot;Apple Color Emoji&quot;, &quot;Segoe UI Emoji&quot;; font-size: 16px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; white-space: normal; background-color: rgb(255, 255, 255); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;">To keep things simple and consistent, Grbl v1.1 has only two reporting options. These are primarily here just for users and developers to help set things up.</p><ul dir="auto" style="box-sizing: border-box; padding-left: 2em; margin-top: 0px; margin-bottom: 16px; color: rgb(31, 35, 40); font-family: -apple-system, BlinkMacSystemFont, &quot;Segoe UI&quot;, &quot;Noto Sans&quot;, Helvetica, Arial, sans-serif, &quot;Apple Color Emoji&quot;, &quot;Segoe UI Emoji&quot;; font-size: 16px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; white-space: normal; background-color: rgb(255, 255, 255); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"><li style="box-sizing: border-box;">Position type may be specified to show either machine position (<code style="box-sizing: border-box; font-family: ui-monospace, SFMono-Regular, &quot;SF Mono&quot;, Menlo, Consolas, &quot;Liberation Mono&quot;, monospace; font-size: 13.6px; padding: 0.2em 0.4em; margin: 0px; white-space: break-spaces; background-color: var(--bgColor-neutral-muted, var(--color-neutral-muted)); border-radius: 6px;">MPos:</code>) or work position (<code style="box-sizing: border-box; font-family: ui-monospace, SFMono-Regular, &quot;SF Mono&quot;, Menlo, Consolas, &quot;Liberation Mono&quot;, monospace; font-size: 13.6px; padding: 0.2em 0.4em; margin: 0px; white-space: break-spaces; background-color: var(--bgColor-neutral-muted, var(--color-neutral-muted)); border-radius: 6px;">WPos:</code>), but no longer bo [... cut ...] ace, SFMono-Regular, &quot;SF Mono&quot;, Menlo, Consolas, &quot;Liberation Mono&quot;, monospace; font-size: 13.6px; padding: 0.2em 0.4em; margin: 0px; white-space: break-spaces; background-color: var(--bgColor-neutral-muted, var(--color-neutral-muted)); border-radius: 6px;">$10=2</code>.</p>

Report Type | Value | Description
-- | -- | --
Position Type | 1 | Enabled MPos:. Disabled WPos:.
Buffer Data | 2 | Enabled Buf: field appears with planner and serial RX available buffer.

</body>
</html>

closed 2025-09-07 by github-actions[bot]