Denvi/Candle #585: heightmap

kusacius, 2022-12-02:
**Versions**
 - Candle version: 1.2.9b

**PC info**
- OS: windows 7

**Describe the bug**
when checking "use heightmap", g-code is breaking.

**To Reproduce**
Steps to reproduce the behavior:
1. Open g-code "1_skyles.ngc" or "Panele4Test_viskas_be_borderio.ngc"
2. Open heightmap file "panele.map"
3. Click on 'use heghtmap'
4. See error

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]
[image]
[image]
[image]

**Example files**
Add G-code files that cause problems.
[test.zip](https://github.com/Denvi/Candle/files/10142858/test.zip)

**Additional context**
strange interpreting arcs in g-code
tested on same other candle versions, problem the same.

mar0x, 2022-12-02:
Hello,
1. Could you please share the heightmap file (when in heightmap "Edit mode", choose "File"->"Save");
2. I'm unable to reproduce the issue with my heightmap on the version I'm currently using: https://github.com/mar0x/Candle/releases/tag/v1.1.9 - please try;

kusacius, 2022-12-03:
[test1.zip](https://github.com/Denvi/Candle/files/10145229/test1.zip)
thank you for quick response, and sorry for my english, i  am from not existing ussr, and my russian is better. i send wrong zip file, this one is right. and i worked on my problem, and solved with correcting gcode syntax. my gcode is generated from altium dxf file to gcode with dxf2gcode utility, i corrected converted gcode files, and attacihng them in next message. i dont know, without heighmap these files are working. so syntax may be is normal. but i maked changes i these files, and they now is working with heigtmap

kusacius, 2022-12-03:
[test2.zip](https://github.com/Denvi/Candle/files/10145272/test2.zip)
there are my changed previous files, and now they working. so i dont know where the problem, in candle heightmap or in dxf2gcode syntax.

closed 2022-12-07 by kusacius