Denvi/Candle #589: Height map is always 0

mattymactutor, 2023-01-16:
I tried searching the issues here and other forums on the web, I apologize if I missed this.

I had a generic CNC 3018 and the heightmap worked perfectly. I just upgrade to the Sainsmart 4040 Pro, and now my heightmap is always 0 at every point. 

Probe Command - G21 G91 G38.2 Z-30 F100; G0 Z0.5; G38.2 Z-2F10

I'm following the same process I did before
- Press Z-Probe to zero
- Press Zero Z to lock in that height
- Create height map
- Probe grid 3x3
- Press Probe
It goes through all of the points but it always returns 0 for all of the heights

Let me know if you need any more info

mattymactutor, 2023-01-17:
I also just tried Candle 1.2.13b and I'm having the same issue.

mar0x, 2023-01-17:
Hello,
Just out of curiosity, it would be good to know the grbl firmware version at the 4040 Pro.

For the first glance, the only difference which may affect the probe function is $21 (hard limits,bool). I would suggest to disable it.

mattymactutor, 2023-01-17:
The GRBL version for both printers is the same here is the output:
```
Grbl 1.1f ['$' for help]
Grbl for GD32
Version:GD32 V2.1
```

I tried changing $21 value to 0 but no luck. 

Re-ran the process with the old CNC just to make sure I'm not doing something stupid and the height map still works with the same process. I'm going to include the full printout after each printer finishes the height map. The 4040 pro has a much slower Z axis speed when performing the height map. The only difference I see in the config settings above is that the axis acceleration's are 30mm/s on the old one and 20mm/s on the 4040 but I can't see that changing much.

What is the process here, Candle asks for a probe at a spot and the CNC firmware completes the probe and sends a signal when finished? Then Candle saves the current height into the height map?

Old CNC Height Map GRBL Output
```
[G54:0.000,0.000,0.000]
[G55:0.000,0.000,0.000]
[G56:0.000,0.000,0.000]
[G57:0.000,0.000,0.000]
[G58:0.000,0.000,0.000]
[G59:0.000,0.000,0.000]
[G28:0.000,0.000,0.000]
[G30:0.000,0.000,0.000]
[G92:-5.000,-19.000,-10.490]
[TLO:0.000]
[PRB:-5.000,-19.000,-10.481:1]
ok
G90 < ok
$G < [GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F4 S3600]
ok
```

4040 PRO Height Map GRBL Output
```
[G54:0.000,0.000,0.000,0.000]
[G55:0.000,0.000,0.000,0.000]
[G56:0.000,0.000,0.000,0.000]
[G57:0.000,0.000,0.000,0.000]
[G58:0.000,0.000,0.000,0.000]
[G59:0.000,0.000,0.000,0.000]
[G28:0.000,0.000,0.000,0.000]
[G30:0.000,0.000,0.000,0.000]
[G92:-27.000,-11.000,-0.801,0.000]
[TLO:0.000]
[PRB:-27.000,-11.000,-0.786,0.000:1]
ok
$J=G21G91X0Y0Z1F2000 < ok
```

mattymactutor, 2023-01-17:
I just noticed that the 4040 pro sends back 4 comma separated values in the [PRB] and the old one sends back 3.

mar0x, 2023-01-17:
Candle [parses PRB response](https://github.com/Denvi/Candle/blob/master/src/frmmain.cpp#L1174) and expect 3 double values. I admin this can be the root of the issue. You may try to change the regexp to match 3rd value and ignore the rest.

mattymactutor, 2023-01-17:
Great! I got it working on Linux by using this line instead:
QRegExp rx(".*PRB:([^,]*),([^,]*),([^]^:^,]*)");

I'd prefer to set this up on a windows machine so now I'm in the process of trying to rebuild from source on Windows. Right now I get error no module for winextras so it seems I have to find the specific 5.4.2 qt version to compile

mattymactutor, 2023-01-17:
Okay so I had to install Qt Creator 5.15 and open the Candle.pro and comment out "QMAKE_LFLAGS += "-Wl,--large-address-aware" in order to build. Once I built, I had to copy the dlls used and plugin folder from Qt install source so I can run from .exe in a dedicated folder. It is now doing the height map correctly giving back the coordinates!! However, it is very very slow in moving the Z-axis. 

Is this based on my GCODE for the probe in the settings? Where can I change the Z speed while height map it being calculated?

mattymactutor, 2023-01-18:
I found it. I think I'm good, thanks for your help. What's your venmo or cashapp?

mar0x, 2023-01-18:
I'm glad you've fixed it!

> What's your venmo or cashapp?

I suppose this is the question for @Denvi .

actools13, 2023-06-24:
ciao e se faccio tutta la procedura e il mandrino continua a girare?

closed 2025-09-06 by github-actions[bot]