Denvi/Candle #591: jog control not working ver; 1.2.15b

jzj3xz, 2023-01-22:
Hi all,
I am running a 3018 Pro and jogging functions work when using UGS and GRBLControl,
but not with Candle ver. 1.2.15b.
Below is the console output:

$J=G21G91X0.000Y5.000Z0.000F2000 < error: Bad number format
$J=G21G91X-5.000Y0.000Z0.000F2000 < error: Bad number format
$J=G21G91X0.000Y-5.000Z0.000F2000 < error: Bad number format
$J=G21G91X5.000Y0.000Z0.000F2000 < error: Bad number format
$J=G21G91X0.000Y0.000Z-5.000F2000 < error: Bad number format
$J=G21G91X0.000Y0.000Z5.000F2000 < error: Bad number format
Is this a known issue and how can I fix this?

mar0x, 2023-01-22:
Hello,
You may try to find similar issues: https://github.com/Denvi/Candle/issues?q=is%3Aissue+Bad+number+format

What is the GRBL version running on your CNC?

jzj3xz, 2023-01-22:
The controller should be on ver 1.1
I did try to run the firmware upgrade but it appears to hang.

closed 2025-09-06 by github-actions[bot]