Denvi/Candle #593: Manual Control of Z-axis - improper response -

SWithnell, 2023-02-05:
**Versions**
 - Candle version: 1.2.13b
 - Firmware version: Unknown

**PC info**
- OS: Windows 10 updates - current - 

**Describe the bug**
I'm using Candle to drive the CNC3-3018PRO small engraving machine to make pcb's.    Just made another this morning all works fine, but a new issue (to me) has emerged.

With Candle set to 'continuous' feed, if I touch the keypad (the touchpad on the laptop) to manually move the spindle up or down,  then the spindle will not stop and crashes the head at either end of travel.   This does not happen in step mode, when it behaves perfectly normally.   Not sure if this is a windows/candle conflict or a candle/grbl controller issue.

Any thoughts appreciated.    Candle works perfectly well other than this specific.

ondras12345, 2023-03-12:
What is you jogging feed rate set to? I'm unable to test at this moment, but I remember having a similar problem when I set it too high.

SWithnell, 2023-03-12:
It is set < than 5mm steps which is well within normal range.    I'll have a think about other settings which might be the trigger.   I've done loads of jobs without an issue, so something has triggered it.   There are two changes one is settings within Candle and the other is a Windows 10 upgrade to latest bug fix.  Those are the only things that change in the environment.

ondras12345, 2023-03-12:
> It is set < than 5mm steps

I'm not sure what you mean by this.
There are two settings in the Jog panel: Step (stored as `jogStep` in `settings.ini`) and Feed (`jogFeed`). If you are having problems with `jogStep=Continuously`, it might be because `jogFeed` is set too high (say 1000 mm/min). Try setting it to 50 and see if that fixes your problem.

SWithnell, 2023-03-12:
Understood.  I was commenting on the manual setting on the control panel - its currently set to 5mm jog step and jog feed at 500.  If I change jog step to continuous.  A tap on the keypad on the z feed causes the head crash.  I'll reduce jog feed to 50 as you suggest as see if that is the issue.

SWithnell, 2023-03-12:
Well its fixed itself.   Just tried your settings and it worked fine.   Increased to 100, fine, 500 fine.   So whatever it was has gone away for now.   I could demonstrate the problem on two different Windows 10 machines and it seems to have gone the same way as it turned up!   Thanks anyway.

closed 2025-09-06 by github-actions[bot]