Denvi/Candle #601: This application failed to start because it could not find or load the Qt platform plugin "xcb".

DylanMS, 2023-03-16:
**Versions**
 - Candle version:  1.1.7

**PC info**
- OS: Linux Ubuntu Studio 22.10

**Describe the bug**
After runing in console showing error 
./Candle 
This application failed to start because it could not find or load the Qt platform plugin "xcb".

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, xcb.

Reinstalling the application may fix this problem.
Neúspěšně ukončen (SIGABRT) (core dumped [obraz paměti uložen]) 

also tryed sudo
sudo ./Candle 
[sudo] heslo pro dylan: 
./Candle: error while loading shared libraries: libQt5OpenGL.so.5: cannot open shared object file: No such file or directory

**Additional context**
Tryed everithing. 
IM DESPERATE

103131, 2023-04-01:
Hi everyone,
I don't have a solution. But I have the same problem.
Do someone have a solution?
Thanks a lot for any advice.

spjt, 2023-04-01:
I got it to work by building the application from source.
There don't seem to be any build instructions on the readme, this is an approximation of what I did while wading through trying to get it to work:

apt install cmake libqt5serialport5-dev xcb xcb-proto
cd Candle
cmake
make
./Candle

103131, 2023-04-02:
Thanks a lot spjt.
I'll give it a try about what you explain.
Thanks a lot for answering me.
Have a nice day.

Barb1211, 2023-04-25:
apt install cmake libqt5serialport5-dev xcb xcb-proto
cd Candle
cmake
make
./Candle

how do I do this. when i try to start program the error message says Not Connected. I have followed all startup instructions. What did I miss?

stephan57160, 2023-05-03:
Had the same troubles
`This application failed to start because it could not find or load the Qt platform plugin "xcb".`

On my side (Ubuntu 22.04), the problem comes from libraries coming with the .tar.gz:
`
libs/libQt5*
`
They are probably generated for a different distro, where xcb is not at the same place.
Would be better to remove them from the tarball and improve the documentation to install Candle on Linux...

So, after I removed Qt5 libraries (and only Qt5 libs), I had to install a few things for i386.
Then I managed to have Candle working with something like below:
```
$ cd Candle
$ rm -rf libs/libQt5*
$ sudo apt install libqt5serialport5-dev:i386
$ # Add other required libs, if needed, but do not forget the ':i386' !
$ ./Candle
```

To rebuild Candle, my method is taken from QT documentations:
```
$ cd Candle/src
$ qmake
$ make
```
Like this, you rebuild according to your PC capabilities (32 or 64 bits).

I still need to check if I still have the "Not Connected" problem, but too late to check that now.

pogo555, 2023-08-24:
Thankyou @stephan57160. Deleting the libQt5 and installing libqt5serialport5-dev:i386 solved my problem.

For those following, the program simply refused to start when double clicked. Running it in a terminal returned command not found "Candle".

closed 2025-09-06 by github-actions[bot]