Denvi/Candle #605: Candle freeze in Run

OasisDweller, 2023-05-21:
Hello,
I'm not sure why Candle freezes on this  one file, it says it's loading but taking forever and not doing anything. Checked and rechecked the file but I can't see the problem. It also says a simple cutout will take 5 minutes???

mar0x, 2023-05-21:
Hello,
Please share the file and the information about your setup. Without details the only correct answer is "Most likely Candle is unable to parse this one file, this is it." :)

It seems 5 minutes cutout estimated for the other file which was parsed successfully...

OasisDweller, 2023-05-21:
Hi, thank you for your time. I use a decent computer to run a Genmitsu 4030
Pro. I use Carveco to create the code and this was a simple rectangle but
I'm sure I am doing something wrong but at a loss to know what.
Joel