Denvi/Candle #606: Home key doesn’t work

Amrt0623, 2023-05-21:
**Versions**
 - Candle version: [e.g.: 1.0.11, 1.1.7, 1.2.9b]
 - Firmware version: [e.g.: 0.9j, 1.1h]

**PC info**
- OS: …
- OpenGL version: ...

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Example files**
Add G-code files that cause problems.

**Additional context**
Add any other context about the problem here.

ondras12345, 2023-05-22:
|ID|Alarm Code Description|
|--|--|
| 8 | Homing fail. Cycle failed to clear limit switch when pulling off. Try increasing pull-off setting or check wiring. |

https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface#alarm-message

Amrt0623, 2023-05-22:
Thanks. 
i checked the limit switches. Xyz have indicator lights on the board.
When i activate any of the limit switches the correct light on the board changes to on.  That is what is supposed to happen.

Amrt0623, 2023-05-22:
So i toggled $5 from 0 to 1 and back.  I had the $21=0
Now the homing cycle works
  But of course the limit switches didn’t work
I set $21 back to 1 and now all works

I had increased the pull of from 3 to 5.  But until i toggled $5 back and forth that didn’t help

the problem seems to be something in the software somwhere?

hoping it never happens again

Amrt0623, 2023-05-22:
Damn. Massivly screw up again

Amrt0623, 2023-05-22:
Finally.  It was picking up different settings files.   

Serious deleting reload reset go

closed 2023-05-22 by Amrt0623