Denvi/Candle #607: Sending File after Height Map

kturunen, 2023-05-27:
**Versions**
 - Candle version: [1.1.7

**PC info**
- OS:Windows 10

**Describe the bug**
[image]

Experienced a broken PCB after making a HeightMap
The last command after HeightMap was Send File 
Why does Z  NOT move up before returning to Home Position?

mar0x, 2023-05-28:
Hello,
How far up it should move? 3 mm? 30 mm? 50 mm?

kturunen, 2023-05-28:
More than heighest point in the height map.

mar0x, 2023-05-28:
I would suggest to add initial movement to safe position in your gcode.

kturunen, 2023-05-29:
After redoing this, I have to wonder how I managed to scratch my fr4. The safe position was set high enough and HeightMapping does return to safe. Maybe I was careless..
This may be closed. Thanks all.

closed 2023-05-29 by kturunen