Denvi/Candle #610: Is the project alive?

ashep, 2023-06-18:
It seems there are no activity on the project according to commits. Is the project alive and has a maintainer?

mar0x, 2023-06-18:
Hello,
The answer depends on your expectations from alive project.
If you have an issue using Candle or need an advise, there is a chance you can get an answer here.
If you need a new feature or improvement to be implemented, it makes sense to scan forks or try to do it yourself.

ashep, 2023-06-18:
Thank you for the answer. Just wondering why activity is so little here, did the author abandon the project?

FrankHovis, 2023-08-01:
Looks to me like its abandoned.

mar0x, 2023-08-02:
Here is "alive" forks/branch ordered by last commit time:

2023-07-31T23:54:56Z [mar0x/Candle/Experimental-fixes](https://github.com/mar0x/Candle/tree/Experimental-fixes)
2023-07-30T23:45:39Z [mar0x/Candle/fixes](https://github.com/mar0x/Candle/tree/fixes)
2023-07-29T19:52:19Z [AdamK2003/Candle/master](https://github.com/AdamK2003/Candle/tree/master)
2023-07-28T00:28:00Z [elhernes/Candle/elh-development](https://github.com/elhernes/Candle/tree/elh-development)
2023-06-20T04:33:58Z [pflagerd/Candle/Experimental](https://github.com/pflagerd/Candle/tree/Experimental)
2023-06-08T21:23:22Z [trcwm/CandleM/master](https://github.com/trcwm/CandleM/tree/master)
2023-06-08T21:11:26Z [trcwm/CandleM/refactoring](https://github.com/trcwm/CandleM/tree/refactoring)
2023-05-03T02:44:04Z [elhernes/Candle/master](https://github.com/elhernes/Candle/tree/master)
2023-04-29T20:09:56Z [pflagerd/Candle/generated_cpp_experiment](https://github.com/pflagerd/Candle/tree/generated_cpp_experiment)
2023-04-28T02:44:58Z [odaki/Candle/macOSbuild](https://github.com/odaki/Candle/tree/macOSbuild)
2023-04-27T19:18:35Z [pflagerd/Candle/master](https://github.com/pflagerd/Candle/tree/master)
2023-04-25T19:14:36Z [elhernes/Candle/macros](https://github.com/elhernes/Candle/tree/macros)
2023-04-16T03:32:21Z [elhernes/Candle/elh-experimental](https://github.com/elhernes/Candle/tree/elh-experimental)
2023-04-15T21:01:48Z [elhernes/Candle/whb04b](https://github.com/elhernes/Candle/tree/whb04b)
2023-03-13T16:27:49Z [stribor/Candle/usestdstrings](https://github.com/stribor/Candle/tree/usestdstrings)
2023-03-12T15:59:55Z [stribor/Candle/master](https://github.com/stribor/Candle/tree/master)
2023-02-10T10:36:36Z [p-Shahpouri/Candle/feature/add-gcode-extention](https://github.com/p-Shahpouri/Candle/tree/feature/add-gcode-extention)
2022-11-24T08:32:07Z [XRay3D/Candle/Experimental](https://github.com/XRay3D/Candle/tree/Experimental)
2022-10-11T19:53:46Z [CaeptnCrunch/Candle/master](https://github.com/CaeptnCrunch/Candle/tree/master)
2022-09-25T10:22:03Z [mamphis/Candle/master](https://github.com/mamphis/Candle/tree/master)

Hope it will help to find a needle in a haystack.

kossnikita, 2023-08-19:
@mar0x could you open issues in your fork?

mar0x, 2023-08-19:
> @mar0x could you open issues in your fork? I'm trying to build your fork on windows, but I always get some kind of error.

Sure. https://github.com/mar0x/Candle/issues
If the error reproduced in main repo, I would suggest to open issue for main repo.

closed 2024-09-07 by ashep