Denvi/Candle #611: Need help to build Experimental Version for Windows 10 using Qt Creator

wamad, 2023-06-22:
Hi all, 

I trying to compile/build the experimental version of Candle using Qt Creator on Win10, but no success. 
I try to build the candle.pro that is in "/src/candle/candle.pro".

The error:
"Cannot find -lcustomwidgets"

Qt Version:
Qt 5.6.1 MinGW 32 bit

Compiler Version:
MinGW 4.9.2 32 bit

Any help will be appreciated, thank you.

[image]

mar0x, 2023-07-09:
Hello,
It seems the issue caused by wrong library path specified in [candle.pro](https://github.com/Denvi/Candle/blob/Experimental/src/candle/candle.pro#L107) and [qtscript_custom.pro](https://github.com/Denvi/Candle/blob/Experimental/src/scriptbindings/qtbindings/qtscript_custom/qtscript_custom.pro#L7).

I would suggest to try the fork with number of fixes - https://github.com/mar0x/Candle/tree/Experimental-fixes - I've tested this branch against Qt 5.12.12 MinGW 32-bit.

wamad, 2023-07-10:
Hi @mar0x, 

Thank you so much for your response. I was able to build using Qt 5.12.12 MingGW 32-bit and run it perfectly.

However, the program cannot load the plugins. Any help on that ?

closed 2023-07-10 by wamad