Denvi/Candle #612: Z Probe for height map for 4 axis controller issue

Kessica, 2023-07-12:
**Versions**
 - Candle version: 1.1.7, 1.2.9b
 - Firmware version:  1.1h

**PC info**
- OS: Windows

When using a 4 axis controller, in general everything works just fine, however when probing to create a heightmap, the controller returns PRB: xPos, yPos, zPos, y2Pos which makes sense. The parser however reads x and y correctly, but then uses the last value for the z position, which is incorrect. This should be a very easy fix, and I would be happy to this myself. However the current source files reference an old version of Qt and therefore references to the serial port have either been removed or moved elsewhere.

I am not familiar with Qt, but a competent c++ developer. If  some kind person could send me the src for 1.17 together with the header files and libraries for the right version of Qt, I would be happy to make a local build here using VS2019. I need heightmaps as I am milling PC boards and it doesnt matter how you clamp them down, they will always bow somewhere, and when working down to 0.01mm, the heights are really needed as V bits depend on the height to get the width of the cut. Cut too deep and tracks dissapear.

Can anyone help me out here? Cheers, Kess

mar0x, 2023-07-12:
Hello,
Please have a look at the [fork](https://github.com/mar0x/Candle). Branch "[fixes](https://github.com/mar0x/Candle/tree/fixes)" is the fixed "master" and "[Experimental-fixes](https://github.com/mar0x/Candle/tree/Experimental-fixes)" - fixed "Experimental" branch.
Both branches contains small fixes to build Candle with Qt 5.15.

Feel free to report any issues with these branches.

Kessica, 2023-07-12:
Thanks for the fast response. The issue occurs with 1.7 and experimental. My issue is with the older versions of Qt. This is what I am really after. The libs and the header files for Qt, as explained above. These are no longer available on the Qt website and any links to archives return with 404. Cheers, Kess

mar0x, 2023-07-12:
Use Qt 5.15. It is still available.

mar0x, 2023-07-12:
This issue looks similar to #589

Kessica, 2023-07-12:
Thanks, yes I did see this thread, hence me going down the route of making a change here locally. The Qt website just times out on the link to 5.5.10 and produces a 404 Page Not found. Perhaps there is an issue with their web site. Ill try again tomorrow. Cheers, Kess

mar0x, 2023-07-12:
> times out on the link to 5.5.10

Have you tried to get Qt 5.15.x instead?

Kessica, 2023-07-12:
Yes, most links take about 8 mins to inform me of a timeout or 404 which is really painful. Having said that, one of links actually came through, which makes me hopeful of getting what I need tomorrow. Thanks for your help Max, really appreciate it :D

closed 2023-07-12 by Kessica