Denvi/Candle #613: error: redefinition of ‘struct QMetaTypeId<QSocketNotifier::Type>’

wryan67, 2023-07-21:
**Versions**
 - Candle version: 1.2.15b
 - Firmware version: 1.1

**PC info**
- OS: Ubuntu 22.04
- OpenGL version: 4.5

**Describe the bug**
During compile of Candle version 1.2.15b, we're getting an error compiling qtscript_QSocketNotifier.cpp:

**To Reproduce**
Steps to reproduce the behavior:
1. check out Experimental branch
2. Run qmake
3. remove the word "relase" from the LIBS customwidgestplugin/release path
4. Run make
5. see error 

**Expected behavior**
Obviously would like to have a clean compile happen.

**Example compile log showing the error**

g++ -c -pipe -O2 -Wall -Wextra -D_REENTRANT -fPIC -DQT_NO_DEBUG -DQT_PLUGIN -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_SCRIPT_LIB -DQT_CONCURRENT_LIB -DQT_CORE_LIB -I. -I. -Iinclude -I/usr/include/x86_64-linux-gnu/qt5 -I/usr/include/x86_64-linux-gnu/qt5/QtWidgets -I/usr/include/x86_64-linux-gnu/qt5/QtGui -I/usr/include/x86_64-linux-gnu/qt5/QtScript -I/usr/include/x86_64-linux-gnu/qt5/QtConcurrent -I/usr/include/x86_64-linux-gnu/qt5/QtCore -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++ -o qtscript_QSocketNotifier.o ../../generated_cpp/com_trolltech_qt_core/qtscript_QSocketNotifier.cpp
In file included from /usr/include/x86_64-linux-gnu/qt5/QtScript/qscriptengine.h:43,
                 from /usr/include/x86_64-linux-gnu/qt5/QtScript/QScriptEngine:1,
                 from ../../generated_cpp/com_trolltech_qt_core/qtscript_QSocketNotifier.cpp:1:
/usr/include/x86_64-linux-gnu/qt5/QtCore/qmetatype.h:2060:12: error: redefinition of ‘struct QMetaTypeId<QSocketNotifier::Type>’
 2060 |     struct QMetaTypeId< TYPE >                                          \
      |            ^~~~~~~~~~~~~~~~~~~

mar0x, 2023-07-22:
Hello,
For Qt >5.15 you may need to apply some of [these changes](https://github.com/mar0x/Candle/commit/f18f06240d1fd96760e52307b46b2b3c75b2895d) or use [Experimental-fixes branch](https://github.com/mar0x/Candle/tree/Experimental-fixes) if you don't want to cherry-pick changes.

wryan67, 2023-07-23:
Thank you so, so much!  That was a great help!   I have it compiled on Ubuntu now.  FYI, here is a list of packages I had to install in case anyone needs them:

```
$ grep DES /etc/lsb-release 
DISTRIB_DESCRIPTION="Ubuntu 22.04.2 LTS"

$ sudo apt update
$ sudo apt upgrade -y
sudo apt install \
  g++ qt5-qmake qttools5-dev qttools5-dev-tools \
  libqt5serialport5-dev libqt5opengl5-dev \
  build-essential automake autoconf autoconf-archive \
  libtool libboost-program-options-dev \
  libgtkmm-2.4-dev gerbv git librsvg2-dev qtmultimedia5-dev \
  libqt5multimediawidgets5 libqt5multimedia5-plugins libqt5multimedia5 \
  qttools5-dev qttools5-dev-tools qtscript5-dev 

```

wryan67, 2023-07-30:
I'm now trying to get candle running on a raspberry pi 4, running ubuntu desktop 22.04 (jammy).  I was thinking I needed an environment variable, but not sure.  When not using any variables, candle starts, but the visualizer is blank.  When I set the QT_QPA_PLATFORM variable to eglfs, I get the following error, any ideas on how to resolve this issue?

```
$ export QT_QPA_PLATFORM=eglfs
$ ./candle 
Unable to query physical screen size, defaulting to 100 dpi.
To override, set QT_QPA_EGLFS_PHYSICAL_WIDTH and QT_QPA_EGLFS_PHYSICAL_HEIGHT (in millimeters).
style "Windows"
style "Fusion"
EGLFS: OpenGL windows cannot be mixed with others.
Aborted (core dumped)
```
System info:
```
$ cat /proc/cpuinfo  | grep Model
Model		: Raspberry Pi 4 Model B Rev 1.4

$ glxinfo | grep -i "opengl version"
OpenGL version string: 2.1 Mesa 22.2.5-0ubuntu0.1~22.04.3

$ cat /etc/lsb-release 
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=22.04
DISTRIB_CODENAME=jammy
DISTRIB_DESCRIPTION="Ubuntu 22.04.2 LTS"
```

mar0x, 2023-07-30:
I have no experience with RPi 4 + jammy + Candle 1.2b yet, but I've successfully built and run [Candle-1.1 with fixes](https://github.com/mar0x/Candle/tree/fixes) on RPi 3 B + Raspbian (bullseye). Here it is [Candle-1.1.9-armhf.AppImage](https://github.com/mar0x/Candle/releases/download/v1.1.9/Candle-1.1.9-armhf.AppImage).

closed 2025-09-06 by github-actions[bot]