Denvi/Candle #617: Can compile Candle  but no button images at runtime on other machine

PaulHosh, 2023-09-07:
Hi Forum.

Many thanks for the Candle program. 

I have compiled Candle 1.1.8 using QT Creator IDE 11.0.2.  I set up QT to use the MIngGW 5.12.12 compiler (available in the Archives under custom install). The program correctly shows the icon images (see link below) that are listed in the qrc file when running in the IDE. 

However, the compiled program does not show the button images when running on another machine.

Is there a required prefix for the qrc file?  Using the "/" or nothing does not show the button images when running Candle on the other machine. I executed qmake and made the release version. No button images visible on the other machine.

It seems that the button images are not embedded into the exe file. How is this done?

Snapshot in QT creator
[image]

Snapshot showing the output. Left image from within QT creator. Right image from other machine.
[image]

thanks
paul

omtim25, 2023-09-08:
tray to copy setting.ini from other machine to candle inside qt creator (located in the same folder of Cancle.exe)

mar0x, 2023-09-08:
Hello,
To be able to run compiled Qt application on other PC, you have to prepare it properly: build "Release" version, copy translation files and "deploy" required Qt libraries and plugins into application directory (so you don't have to install Qt to run Candle on other PC).
I don't know how to do it from Qt Creator. There is a command line tool "windeployqt" to perform the last step.