Denvi/Candle #618: Visualizer not showing progress

Project-Man-2023, 2023-09-24:
**Versions**
 - Candle version: [e.g.: 1.0.11, 1.1.7, 1.2.9b]
 - Firmware version: [e.g.: 0.9j, 1.1h]

**PC info**
- OS: …
- OpenGL version: ...

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Example files**
Add G-code files that cause problems.

**Additional context**
Add any other context about the problem here.

Project-Man-2023, 2023-09-24:
I just purchased a Sainsmart Prover XL 6050.
I previously used Candle on a 3018 machine with no issue.
When I run a project on the 6050 the Visualizer stays idle, shows no progress and run clock doesn’t start while the job is rinning on the cnc. When I reach the end of the project the spindle turns off, z axis raises but I never receive the job ended popup. The machine seems to be in idle. Pause and abort buttons won’t work. The only way to end is with the reset button. Any idea what is going on? Thanks

mar0x, 2023-09-25:
Hello,
Could you please share the following information:
Candle version, GRBL version and GRBL settings.
Thank you.

Project-Man-2023, 2023-09-25:
Candle version 1.1.7
I don't know how to find the firmware version
the GRBL version is what is coming from Autodesk Fusion 360
[image]

My Issue: When I run a project on the 6050 the Visualizer stays idle, shows no progress and run clock doesn’t start while the job is rinning on the cnc. When I reach the end of the project the spindle turns off, z axis raises but I never receive the job ended popup. The machine seems to be in idle. Pause and abort buttons won’t work. The only way to end is with the reset button.

(1001)
(Machine)
(  vendor: Sainsmart)
(  model: ProVer XL 6050)
(  description: This machine has XYZ axis on the Head)
(T4 D=25.4 CR=0 - ZMIN=0 - face mill)
G90 G94
G17
G21
(-Attention- Property Safe Retracts is set to Clearance Height.)
(Ensure the clearance height will clear the part and or fixtures.)
(Raise the Z-axis to a safe height before starting the program.)

(Face2)
T4
S5000 M3
G17 G90 G94
G54
G0 X596.51 Y4.631
Z16
G0 Z6
G1 Z2.54 F333.3
G18 G3 X593.97 Z0 I-2.54 K0 F1000
G1 X580
X0
G17 G2 X0 Y21.783 I0 J8.576
G1 X580
G3 X580 Y38.936 I0 J8.576
G1 X0
G2 X0 Y56.089 I0 J8.576
G1 X580
G3 X580 Y73.242 I0 J8.576
G1 X0
G2 X0 Y90.395 I0 J8.576
G1 X580
G3 X580 Y107.548 I0 J8.576
G1 X0
G2 X0 Y124.7 I0 J8.576
G1 X580
G3 X580 Y141.853 I0 J8.576
G1 X0
G2 X0 Y159.006 I0 J8.576
G1 X580
G3 X580 Y176.159 I0 J8.576
G1 X0
G2 X0 Y193.312 I0 J8.576
G1 X580
G3 X580 Y210.465 I0 J8.576
G1 X0
G2 X0 Y227.617 I0 J8.576
G1 X580
G3 X580 Y244.77 I0 J8.576
G1 X0
G2 X0 Y261.923 I0 J8.576
G1 X580
G3 X580 Y279.076 I0 J8.576
G1 X0
G2 X0 Y296.229 I0 J8.576
G1 X580
G3 X580 Y313.382 I0 J8.576
G1 X0
G2 X0 Y330.535 I0 J8.576
G1 X580
G3 X580 Y347.687 I0 J8.576
G1 X0
G2 X0 Y364.84 I0 J8.576
G1 X580
G3 X580 Y381.993 I0 J8.576
G1 X0
G2 X0 Y399.146 I0 J8.576
G1 X580
G3 X580 Y416.299 I0 J8.576
G1 X0
G2 X0 Y433.452 I0 J8.576
G1 X580
G3 X580 Y450.604 I0 J8.576
G1 X0
G2 X0 Y467.757 I0 J8.576
G1 X580
G18 G2 X582.54 Z2.54 I0 K2.54
G0 Z16
G17
M5
M30

mar0x, 2023-09-25:
Enable "Console/Show UI commands" in Candle preferences and you can see message like this:
```
[CTRL+X] < Grbl 1.1f ['$' for help]
```
in Candle Console view right after you connect the CNC to your PC. Here "Grbl 1.1f" - is a version string.

Type `$$` to the small input field below the Console, hit Enter and you can see something like:
```
$$ < $0=10
$1=254
$2=0
$3=2
...
```
This is your [GRBL configuration](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration), please copy and share it here.

Candle window screenshot with GCode loaded and CNC connected may also help.
Does other Candle features works correctly (Jog, Home, etc.) ?

Project-Man-2023, 2023-09-25:
$32=0
$100=160.000
$101=160.000
$102=160.000
$110=2000.000
$111=2000.000
$112=2000.000
$120=300.000
$121=300.000
$122=300.000
$130=600.000
$131=500.000
$132=110.000
ok

Project-Man-2023, 2023-09-25:
Jog and homing features work

Project-Man-2023, 2023-09-26:
It means nothing happens in the visualizer window. Machine is running but the visualized window shows no progress. Running time clock does not do anything.

Project-Man-2023, 2023-09-26:
It looks like my issue is solved. I contacted Sainsmart and described my issue. They sent me new code to enter into the console. That solved the problem. Machine code was bad out of the box. Thx for working on it.

mar0x, 2023-09-26:
Glad to know it!
It makes sense to share the "code to enter into the console" here and close this issue as resolved.

Project-Man-2023, 2023-09-26:
This is code received from Sainsmart. This solved the problem.

$0=10
$1=25
$2=0
$3=0
$4=0
$5=0
$6=0
$10=1
$11=0.010
$12=0.002
$13=0
$20=0
$21=1
$22=1
$23=0
$24=25.000
$25=500.000
$26=250
$27=3.000
$30=10000
$31=0
$32=0
$100=160.000
$101=160.000
$102=160.000
$110=2000.000
$111=2000.000
$112 = 2000.000
$120=300.000
$121=300.000
$122=300.000
$130=600.000
$131=500.000
$132=110.000

closed 2023-09-26 by Project-Man-2023

drik, 2024-08-07:
Hello, I had the same issue, and when I compared your values with mine, I realized $10 was the difference. Mine was $10=2 when I changed it to $10=1, it worked

Project-Man-2023, 2024-08-07:
👍TimSent from my iPhone

papillion47, 2024-09-27:
[image]

mar0x, 2024-09-27:
@papillion47 Try "Service"->"Settings"->"Set to defaults".

papillion47, 2024-10-03:
Varsayılanlara ayarlayı yaptım. Değişen bir şey olmadı..