Denvi/Candle #619: Probe not working with Height Map and just crashing into the bed?

PigeonFX, 2023-10-22:
Hi, my probe works fine, until I try and use it for a height map. 
Cant use the probe button to find z0 on my copper sheet without issue, but when I try and do a height map it just crashes into the stock.

[image]

[image]

Using a Windows 10 PC

peter-ch, 2023-11-06:
I have the same problem.

mar0x, 2023-11-06:
Hello,
According to screenshot, the connection with CNC was lost. Have you unplugged it intentionally?

What is your CNC GRBL version?
What is your Candle "Heightmap probing feed" setting value?

Does the issue reproduced on [Candle 1.1.7](https://github.com/Denvi/Candle/releases/tag/v1.1) ?
Does the issue reproduced on patched and experimental [Candle 1.1.9](https://github.com/mar0x/Candle/releases/tag/v1.1.9) ?

peter-ch, 2023-11-12:
The problem for me seems to be that I had lifted the probe too high from the bed. To make the heightmap successfully, do this: 
1. Z probe to fine the 0
2. Set the Z=0 point
3. Lift the probe with 2 mm, adjust the Z travel parameters to 2 and -5 mm 
4. Run the heightmap probing

closed 2025-09-06 by github-actions[bot]