Denvi/Candle #620: visualisation issue with black window and refresh issue

FelixCNC, 2023-10-25:
I'm using Candle version: 1.1.7 and1.2.9b on a WIN 7 PC with integrated intel GPU Intel(R) HD  Graphics 520 (latest drivers updated)

when opening Candle, the visaulisation windows shows somehow unrefreshed coordinate system. Background is black, independent of the background color set in the settings.
If NC file is loaded, similar - unrefresehd picture- appears.

Any ideas how to solve the issue?

[image]

son1csun, 2024-06-12:
I have a similar-appearing problem.  Black screen, acts like a transparent window into things happening behind it.  No graphics, no white screen, no visualisation at all if I open a gcode file.  Just a transparent window into what's behind it.

I just updated to Ubuntu 24.04 and re-installed Candle.  This was working for years on this machine, with Ubuntu 18.04.   I have candle working on several other machines in our office (windows and ubuntu).

Here is the current graphics card and driver info:

00:02.0 VGA compatible controller: Intel Corporation Iris Plus Graphics G1 (Ice Lake) (rev 07)
	Subsystem: Hewlett-Packard Company Iris Plus Graphics G1 (Ice Lake)
	Kernel driver in use: i915
	Kernel modules: i915

[image]

First I tried installing with Qt, but that produced this problem.  Then I tried installing with cmake.  No difference.

It will connect to our little CNC engraving machine, but that seems irrelevant, if  we can't even open a gcode file and review it.

I'd appreciate your thoughts and ideas.  The OS upgrade has turned into a massive downgrade for now.

Thanks, son1csun

mar0x, 2024-06-12:
@son1csun, when you run Candle from terminal, it there anything outputs into terminal?

son1csun, 2024-06-13:
@mar0x Here's a screenshot of Candle run from the terminal.  The desktop background is actually shown in the Candle window, but somehow it is displayed as solid black in this screenshot.

[image]

When I start from the terminal, this message is shown:

QSocketNotifier: Can only be used with threads started with QThread
locale: "en"

I'm not sure how to fix this.  I have a german keyboard, but my computer is set to english language.

Perhaps I should just use an earlier version of Candle?

Thanks!

mar0x, 2024-06-13:
The output
```
locale: "en"
```
is harmless and only states the translation language used (see [main.cpp:50](https://github.com/Denvi/Candle/blob/master/src/main.cpp#L50)) .
`QSocketNotifier` message is more interesting, but seems unrelated to your graphics issue.

In my case there also message `shader program created` during GL widget initialization.

son1csun, 2024-06-13:
@mar0x Thanks.  The complete output in the terminal after start is:

QSocketNotifier: Can only be used with threads started with QThread
locale: "en"
shader program created
preparing vectors GcodeDrawer(0x5fa61e1aec20)
lines count 0
preparing vectors GcodeDrawer(0x5fa61e19b9e0)
lines count 0

Not sure if this is helpful.  Tried installing 1.1.7 and am now dealing with a different error there:

./Candle: error while loading shared libraries: libgobject-2.0.so.0: cannot open shared object file: No such file or directory

This seems like a common linux problem that has a solution somewhere online...

mar0x, 2024-06-13:
Have you tried [Candle-1.1.9-Linux-x86_64.AppImage](https://github.com/mar0x/Candle/releases/download/v1.1.9/Candle-1.1.9-Linux-x86_64.AppImage) (download, make it executable and run) ?

son1csun, 2024-06-13:
@mar0x  Just downloaded, built with Qt, and ran Candle-grbl_1_0 from terminal.  Same visualisation issue:

[image]

Text that appears in the terminal:

QSocketNotifier: Can only be used with threads started with QThread
locale: "en"
style "Windows"
style "Fusion"
shader program created
preparing vectors GcodeDrawer(0x584dd5960dc0)
lines count 0
preparing vectors GcodeDrawer(0x584dd59cfcd0)
lines count 0

son1csun, 2024-06-13:
> Have you tried [Candle-1.1.9-Linux-x86_64.AppImage](https://github.com/mar0x/Candle/releases/download/v1.1.9/Candle-1.1.9-Linux-x86_64.AppImage) (download, make it executable and run) ?

@mar0x   That + Service > Settings > Set to defaults did it!  Whew...  Now to delete all the non-working junk...  

Muchas Gracias! Grazie Mille! Danke sehr! etc.

Now back to work

mar0x, 2024-06-13:
@son1csun, I'm glad it works for you. Most likely there was an issue with build environment in your new Ubuntu 24.04 (something missing).

pavelstano, 2025-08-01:
Try this solution https://github.com/Schildkroet/Candle2/issues/62#issuecomment-2356045282

closed 2025-09-16 by github-actions[bot]