Denvi/Candle #626: Heightmap process seems to gradually lower Z flying height during processing until bit touches PCB during XY-move

poes8943, 2024-02-11:
**Versions**
 - Candle version: 1.1.7
 - Firmware version: 1.1f
 - Milling Machine: CNC 1419

**PC info**
- OS: Win10Pro

**Describe the bug**
Heightmap process seems to gradually lower Z flying height during processing until bit touches PCB during XY-move

**To Reproduce**
using standard probe command: G21G91G38.2Z-30F100; G0Z1; G38.2Z-2F10
doing a manual Z-probe first using Z-probe control button followed by a Z-move UP 0.5mm
starting a 3x3 heightmap (20x25mm PCB area) Zt:2.00 Zb:-1.00
automatic results using heightmap feature: 
1.964, nan, nan
1.976, 1.966, 1.559
0.400, 0.784, 1.179
>>> flying height of the probe gets lower after every probe until the probe touches the PCB
>>> heightmap stops after 7 probes with ERROR:9
manual results: (doing the first 3 probes manually with jog and z-probe G command 'G38.2Z-2F10')
Grbl 1.1f ['$' for help]
G38.2Z-2F10 < [PRB:0.000,0.000,0.009:1]
G38.2Z-2F10 < [PRB:10.000,0.000,-0.004:1]
G38.2Z-2F10 < [PRB:20.000,0.000,-0.018:1]
The manual process gives reasonable values ... so stepper motors and spindles seem to be fine and I am not loosing any step counts.
... no ideas why this happens ... any clues or known issues?

**Expected behavior**
There should be no difference between 'manual' and automatic heightmap.

poes8943, 2024-02-11:
further test steps using Z-probe control button:
1) push Z-probe control button and perform a Z-zero after that (work coordinates now at 0.000 0.000 0.000)
2) jog Z-up 0.5mm and pushing Z-probe control button again (Z axis work coordinate at 0.399 now)
3) jog Z-up 0.5mm and pushing Z-probe control button again (Z axis work coordinate at 0.807 now)
4) jog Z-up 0.5mm and pushing Z-probe control button again (Z axis work coordinate at 1.199 now)
... there seems to be roughly an error of 0.4mm after each step which is adding up ...

same test with manual commands:
1) push Z-probe control button and perform a Z-zero after that (work coordinates now at 0.000 0.000 0.000)
2) jog Z-up 0.5mm and executing 'G38.2Z-2F10' (Z axis work coordinate at 0.009 now)
3) jog Z-up 0.5mm and executing 'G38.2Z-2F10' (Z axis work coordinate at 0.006 now)
4) jog Z-up 0.5mm and executing 'G38.2Z-2F10' (Z axis work coordinate at 0.007 now)

mar0x, 2024-02-11:
Hello,
Please follow the long discussion in #550 maybe you find something useful.

poes8943, 2024-02-11:
thanks - was reading #550 some hours ago but seems not to solve my issue (revisiting the thread again)

Hostaler, 2024-02-11:
How can I put all this to work? Thank you very much.

closed 2025-09-05 by github-actions[bot]