Denvi/Candle #629: Grbl Candle tool path problems

hibb2007, 2024-02-20:
I am using Carveco Maker along with Grbl Candle controller for my Foxalien 40-40xe cnc router. As I change step over rates and save them to open in Candle, the time frames are all over the place. It is probably something i am doing wrong as i am fairly new to this. For example, i am trying to do a rather large project. The roughing pass worked well. When i opened the finishing tool path, that's when i started with issues. I am using a 30 degree 1/8" v bit. Initially i had my stepover set to .05, which i have learned since is too much. But it said it was going to take 1 hr 31 minutes. Which is wierd because a similar project the night before took over 6 hours. I changed my stepover to .00125, understanding it will take longer but should get a much nicer finish. Candle then said it would only take 13 minutes. So that made no sense in that it should have taken much longer, instead it cut the time down. So then just to see what happens, i changed the stepover to .0011. That changed the time to 6 hr 47 min. A much more realistic time. But i dont understand how .00125 was 13 minutes and .0011 is 6 hr 47 min. Hoping someone can tell me what i am doing wrong and point me in the right direction. Thanks!

mar0x, 2024-02-21:
Hello,
As far as I understand, the only issue you have is the "Estimated Time" value, calculated by Candle. Correct?
What is "stepover"? How does it affect Candle or gcode?
Could you please share 2 gcode files so I can compare estimations.

Usually, Candle underestimate the time because the estimation based only on line segment length (see [updateProgramEstimatedTime()](https://github.com/Denvi/Candle/blob/master/src/frmmain.cpp#L1792) ) completely ignoring acceleration settings.

hibb2007, 2024-02-22:
Well I don't exactly know what the issue is. I havent tried to run it yet because the estimated times seem unrealistic and i dont want to mess the project up. I did a rough pass with an 1/8 inch endmill and that pass worked perfectly fine. The estimated time and actual run time were reasonably close, within 10 minutes. The project size is 10 in x 14.5 in. So i know the 30 degree v bit is going to take longer. Also since it is the finishing pass, it is doing the toolpath as raster. So it is going to totally cover the whole project and will be lengthy. Stepover is how far the bit is moved over each time the machine takes a pass. and I tried to upload the g codes for you but it didnt seem to let me. Also, i am fairly new to this so it may be something i am doing wrong or a setting i have wrong. I contacted the Carveco support line and it doesnt seem to be a Carveco problem. So it has to be something with grbl candle. Another thing that may or may not be helpful, none of my other projects did this, but this came up with an orange circle and lines somewhat like crosshairs over the project. I am sure it means something but i dont know what. Im going to try to post a screen shot of what i am talking about. I will also try a comparison screenshot of another toolpath of the same project. the only difference is .0011 inch stepover and .00125 stepover. The estimated time are drastically different.
[image]
[image]

closed 2025-09-05 by github-actions[bot]