Denvi/Candle #632: New fork, refactor, does it make sense?

etet100, 2024-04-09:
Hi. I wanted to add some minor functionality, and after reviewing the code I ended up STARTING to rewrite much of it from scratch. But at some stage I came to the conclusion that it might not make sense. 

So a question, mainly to people who have created/create forks of the Candle app themselves. In your opinion, does it make sense to rebuild it completely? The idea is to build it modularly, separate the UI from the logic (to allow easy switching of different UIs) and actually rewrite much of the code. 

I wonder if you would see something in such a project that would make you abandon your own forks in favor of such a refactored version. And most importantly - would you be willing to add something from yourselves to such a version?

barkermn01, 2024-04-21:
Not really, at that point your not forking for a merge it's a complete re-write, and it meaning nothing other people are working on currently can be merged in so it break the point of git.

Best option is to create a new git repo from the start and port the code you need over, and just make sure you licence & README.md files credits this project and your new project is also licenced under GPLv3 as that's a requirement of GPL Licence this software is using.

etet100, 2024-04-21:
Actually, I don't think that's quite what I meant. I never planned to merge it to the original (dead) Candle. It was more about the point of creating a new app at all.  And I know that the work already done will not necessarily be easy to merge into my fork. But still, someone may feel like doing it.

etet100, 2024-04-22:
You know... hundreds of open issues, dozens of PRs. For me, it's more than "suspended." I z tego powodu powstało X forków.. That's why I was wondering if it's worth rewriting it (which I'm already doing) and making sure the project lives.

barkermn01, 2024-04-29:
My whole thing was a make a copy not a fork, because your repo's compatibility with this repos code base will be shot from what you have made sound like a very good idea.