Denvi/Candle #633: Crazy Jog

Juank23, 2024-05-02:
Good morning Denvi, I downloaded the source code version of candle 1.7 and I'm trying to change the functions of the numeric keypad, by the arrows, but when I compile and open the program, I move the machine with the arrows and it goes to one end and stays paid, is there any way to make this change I remain atentro thanks

Juank23, 2024-05-02:
hello mar0x , Thanks for your prompt reply, actually I am using version 1.1.8, I have a small laptop which does not have the numeric keys, , with the mouse it moves fine, but if I use the arrows it goes to one side and I have to restart, I modified the following lines in the code ....
[image]
[image]
[image]

mar0x, 2024-05-02:
Please enable "Show UI Commands in console", reproduce your issue and see/share jog commands in your console.

Juank23, 2024-05-02:
in frmmain.cpp change the numbers by the arrows of the keyboard as I show in the following image, and when pressing any key of the arrows the machine moves without stopping, but with the mouse if they work well but after closing and opening candle
[image]
[image]