Denvi/Candle #637: Candle 1.2.15b  probe command missing

Patsjemoe, 2024-06-22:
Hi,
When using candle 1.1.17, you have a screen (see attached)
[image]
In this screen, we change the height of the probe, and start probing/height map.
In version 1.2.15b, I cannot find this screen to adapt.
Also the safe position commands is missing.
1) Please confirm where to find this ?
2) is there a user manual for Candle 1.2.15b ?
thanks in advance
best regards
Ludo

Hostaler, 2024-06-22:
[image]

I think what you are looking for is this.
If I'm right, please share the line. 
Thank you.

Patsjemoe, 2024-06-22:
Hi,
This looks to be right, but could not edit it...any suggestion ?
Any idea on how to simply provide a safe position command, instead of the script that is not explained anywhere ?
Thanks
Regards
Ludo

Hostaler, 2024-06-22:
Double click on the line and yes it can be edited.

Patsjemoe, 2024-06-22:
Hi Hostaler,
I can indeed edit ..
But I have not used any cnc before, and only have an explanation for 1.1.17 but need to put it into 1.2.15b.
In 1.1.17, they suggested following line : 
G91G21G38.2Z-50F100; G92 Z12.2; G0Z5M30 where Z12.2 is my probe height.
In 1.2.15b, I have default following : 
G21G91 G38.2Z-30F100 G0Z1 G38.2Z-2F10, and am not sure why it is so different, specially  G0Z5M30 versus G0Z1 G38.2Z-2F10
I was considering to use G91G21G38.2Z-50F100; G92 Z12.2; from 1..17, but what to do with the G0Z5M30 versus G0Z1 G38.2Z-2F10 ?
thanks 
regards
Ludo

Hostaler, 2024-06-22:
Do the trial and error technique. Add the line you had in the previous version if it worked for you. With the M30, the M30 is not going to cause you any problems.
Googled g codes for grbl.

Patsjemoe, 2024-06-23:
OK, thank you for your feedback. Probing works fine, height map too.

closed 2024-06-23 by Patsjemoe

Hostaler, 2024-06-24:
Can you send the lines for the probe and for the height map?