Denvi/Candle #638: candle 1.2.15b M6 and alarm issue

Patsjemoe, 2024-06-24:
[image]
[image]
[image]
[image]

Hi,
I am trying to get my 3018 CNC to work with flatcam and candle, but need help : 
1)In the picture, you can see that I am milling my B-Cu pcb.  Am using a 0.1 mm 20deg V shape tool, but use only 1 tool to mill.
When starting my code, it stops at line 30 with error 20 on code M6 , which is tool change....As I only use 1 tool, I do not understand why he requests to change. I ignore this error, and it continues.
2) when I the push the send button, it gives error code 9, complaining about the alarm being on.
   When I do a reset, it starts from zero again with error   on M6.
   Where can I turn off the alarm, and give a "continue" command ?

Gcode 1 and 2 are attached (could not copy from candle...ppfff)
[image]
[image]

Regards
Ludo

mar0x, 2024-06-24:
Hello,
"M6" command is unsupported by GRBL v1.1 this why you are getting "error:20" message. "Ignore" is safe and correct response for this message.

Second message "error:9" is unrelated to previous one and seems the result of alarm. Please enable "Console"->"Show G-code program commands" (for Candle 1.1.x) and share console output.

Patsjemoe, 2024-06-24:
Hi,
Thank you for your quick reply.

Console :
(G-CODE GENERATED BY FLATCAM V8.994 - WWW.FLATCAM.ORG - VERSION DATE:
2020/11/7) < ok
(NAME: SJOTTERACTUATORESP32-B_CU.GBR_ISO_COMBINED_CNC) < ok
(TYPE: G-CODE FROM GEOMETRY) < ok
(UNITS: MM) < ok
(CREATED ON SUNDAY, 23 JUNE 2024 AT 18:56) < ok
(THIS PREPROCESSOR IS THE DEFAULT PREPROCESSOR USED BY FLATCAM.) < ok
(IT IS MADE TO WORK WITH MACH3 COMPATIBLE MOTION CONTROLLERS.) < ok
(TOOL DIAMETER: 0.1 MM) < ok
(FEEDRATE_XY: 120.0 MM/MIN) < ok
(FEEDRATE_Z: 60.0 MM/MIN) < ok
(FEEDRATE RAPIDS 1500.0 MM/MIN) < ok
(Z_CUT: 0.0 MM) < ok
(Z_MOVE: 2.0 MM) < ok
(Z START: NONE MM) < ok
(Z END: 15.0 MM) < ok
(X,Y END: NONE MM) < ok
(STEPS PER CIRCLE: 64) < ok
(PREPROCESSOR GEOMETRY: DEFAULT) < ok
(X RANGE:   -0.1798 ...   44.0202  MM) < ok
(Y RANGE:    0.3440 ...   75.8205  MM) < ok
(SPINDLE SPEED: 0.0 RPM) < ok
G21 < ok
G90 < ok
G94 < ok
G01 F120.00 < ok
M5 < ok
G00 Z15.0000 < ok
G00 X0.0000 Y0.0000 < ok
T1 < ok
M6 < error:20

[MSG:Check Limits] (repeated a lot)
ok
(G-CODE GENERATED BY FLATCAM V8.994 - WWW.FLATCAM.ORG - VERSION DATE:
2020/11/7) < ok
(NAME: SJOTTERACTUATORESP32-B_CU.GBR_ISO_COMBINED_CNC) < ok
(TYPE: G-CODE FROM GEOMETRY) < ok
(UNITS: MM) < ok
(CREATED ON SUNDAY, 23 JUNE 2024 AT 18:56) < ok
(THIS PREPROCESSOR IS THE DEFAULT PREPROCESSOR USED BY FLATCAM.) < ok
(IT IS MADE TO WORK WITH MACH3 COMPATIBLE MOTION CONTROLLERS.) < [MSG:Check
Limits]
ok
(TOOL DIAMETER: 0.1 MM) < ok
(FEEDRATE_XY: 120.0 MM/MIN) < ok
(FEEDRATE_Z: 60.0 MM/MIN) < ok
(FEEDRATE RAPIDS 1500.0 MM/MIN) < ok
(Z_CUT: 0.0 MM) < ok
(Z_MOVE: 2.0 MM) < ok
(Z START: NONE MM) < ok
(Z END: 15.0 MM) < ok
(X,Y END: NONE MM) < ok
(STEPS PER CIRCLE: 64) < ok
(PREPROCESSOR GEOMETRY: DEFAULT) < ok
(X RANGE:   -0.1798 ...   44.0202  MM) < ok
(Y RANGE:    0.3440 ...   75.8205  MM) < ok
(SPINDLE SPEED: 0.0 RPM) < ok
G21 < ok
G90 < error:9
G94 < error:9
G01 F120.00 < error:9
M5 < error:9
G00 Z15.0000 < error:9
G00 X0.0000 Y0.0000 < error:9
T1 < error:9
M6 < error:9
error:9
[MSG:Check Limits]
ok

mar0x, 2024-06-24:
"Check Limits" - most likely the root of the issue.

Please share your firmware (GRBL) version and all GRBL settings (see https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration).

Patsjemoe, 2024-06-24:
Hi,
Have deleted a couple of lines, including the m6, but also it moved the
drill away, and this created the switches to activate.
It was good to have a starting point to think what was happening..
I still do not understand why flatcam is producing the tool change, as I
use only the V tool.
Thanks for your help.
regards
Ludo

mar0x, 2024-06-24:
Again, "M6" is NOT an issue. Instead of blindly change the g-code here and there is makes sense to understand what is causing the issue and familiarize yourself with tools you are using.

Clear understanding of the GRBL version and configuration is one of the key point to get correct advise. Unfortunately people prefer to share screenshots with error message instead of valuable information asked in the issue template.

If you are happy with the solution you have, please close the issue.

Patsjemoe, 2024-06-24:
Hi, 
I dit not blindly change g-code, but based on your previous info of the limit switches, I further looked into it, and although M6 could just be ignored, the code behind that moved the head too far, so it hit the limit switch. I took this out and it was solved.....
Therefore, your comment of just sending some screenshots and not understand was not really honest. ....it would though help to better understand if the documentation would be more complete....but we do it with what we have.
But anywhay, thank you for pointing me in the right direction.
best regards
ludo

closed 2024-06-24 by Patsjemoe

mar0x, 2024-06-25:
Thank you for sharing the solution!

### Summary
#### Observed
Unexpected error message "**M6 < error:20**" followed by message with list of "**error:9**". CNC stopped.

#### Analysis
The firmware version is unknown. The exact sequence of operations is absent. G-code is partially available. The words `[MSG:Check Limits]` in console output points to guess about limits violation. To understand what limit and why it was triggered the error it requires to get GRBL settings using `$$` and check [Soft](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#20---soft-limits-boolean), [Hard](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#21---hard-limits-boolean) limits and [Travel](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#130-131-132--xyz-max-travel-mm) settings, then check the CNC position at the error state.

### Solution

1. Manually remove "**M6**" line from g-code to avoid  "**M6 < error:20**".
2. Manually remove some (what?) movement to avoid "**error:9**".