Denvi/Candle #639: Tool height sensor.

DjinGri, 2024-07-08:
**Versions**
 - Candle version: 1.1.8
 - 
Normally closed tool height sensor

To determine the height of the cutter, I use a normally closed limit switch. The spark plug works flawlessly. 
 The problem occurs when creating a height map. When started, the cutter touches the surface of the material, moves to the set position and goes into “Emergency” mode.

Configuration

 Dipstick/probe:
 probe:                           
  pin: gpio.36:high:pd

mar0x, 2024-07-08:
Hello,
From the description it is not clear for me what is the issue. The height map creation process is based on probe (or cutter) touch the surface. Candle is unaware of pins and levels. It "talks" to CNC via GRBL protocol.

1. What is your CNC GRBL version?
2. Does the probe [image] function works for your? Please share probe command value (from Candle settings).
3. I'm sorry, I don't understand the purpose of the spark plug despite the fact it works. Please explain.
4. Please enable "Settings"->"Console"->"Show UI commands" and share console output during probe.
5. Check/set the "Heightmap probing feed" value.

DjinGri, 2024-07-10:
FluidNC firmware. 

Probe function works. The video shows: 1. launching the “probe” function and how it works (console).
2. Launch the “height map” function (console).

The height map generates an error only when a normally closed limit switch is connected. When connecting a normally open limit switch, the height map is built normally, that is, without problems.[https://drive.google.com/file/d/1crFSECORisDV7RSRy-WpfFgu0msTqj3f/view?usp=sharing](url)

DjinGri, 2024-07-10:
[](url)

DjinGri, 2024-07-10:
[
[image]
[image]
](url)

mar0x, 2024-07-10:
I'm sorry, I'm unable to review the video, guess what is going on there and try to read the console output. If you still need my help please answer all my questions above in text.

DjinGri, 2024-07-10:
Fluid NC firmware

Function "Probe sensor running ES"
--------------------------------------------
Function "Height map does not work NO"
Console

[MSG:ERR: Reset to continue]
[MSG:ERR: Reset to continue]

DjinGri, 2024-07-11:
Probe "Yes"!

G91G21G38.2Z-20F100 < [PRB:0.000,0.000,-4.837:1]
ok
G92 Z1.0 < ok
G0Z5M30 < [MSG:INFO: Program End]
ok
[GC:G1 G54 G17 G21 G90 G94 M5 M9 T0 F100 S1000]
ok
-------------------------------------\

height map "NO"!

[MSG:ERR: Reset to continue]
[MSG:ERR: Reset to continue]

[image]

mar0x, 2024-07-11:
Вопрос: какая версия GRBL?
Ответ: Fluid NC firmware
Комментарий: версия выводится в консоль Candle после подключения или сброса станка. Выглядит вот так: GRBL 1.1f  (это пример)

Вопрос: работает ли функция "Probe"?
Ответ: да

Вопрос: поделитесь командой, которая используется для функции "Probe" (из настроек)
Ответ: команду можно вычленить из лога или увидеть на снимке экрана
Комментарий: немного странная команда, но раз считается, что она работает, то Ок

Вопрос: поясните какую функцию выполняет "spark plug" из описания проблемы
Ответ: -

Вопрос: включите показ команд и поделитесь тем, что выводится при выполнении функции "Probe"
Ответ: задание выполнено успешно:

```
G91G21G38.2Z-20F100 < [PRB:0.000,0.000,-4.837:1]
ok
G92 Z1.0 < ok
G0Z5M30 < [MSG:INFO: Program End]
ok
[GC:G1 G54 G17 G21 G90 G94 M5 M9 T0 F100 S1000]
ok
```

Вопрос: проверьте/установите скорость подачи для тестирования высоты при построении карты высот
Ответ: -

Вопрос: Есть ли смысл задавать новые вопросы?

closed 2025-09-05 by github-actions[bot]