Denvi/Candle #641: backwards

david1560, 2024-07-31:
**Versions**
 - Candle version: [e.g.: 1.0.11, 1.1.7, 1.2.9b]
 - Firmware version: [e.g.: 0.9j, 1.1h]

**PC info**
- OS: …
- OpenGL version: ...

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Example files**
Add G-code files that cause problems.

**Additional context**
Add any other context about the problem here.
when I use candle grbl everything cuts good but when i cut numbers and letters they come out backwards but the controls are the right way when i hit left it goes left and up is up and forward is forward and so on so what am I doing wrong can anyone help me please

david1560, 2024-07-31:
Add any other context about the problem here.
when I use candle grbl everything cuts good but when i cut numbers and letters they come out backwards but the controls are the right way when i hit left it goes left and up is up and forward is forward and so on so what am I doing wrong can anyone help me pleas

mar0x, 2024-07-31:
Hello,
Candle visualize tool path for you GRBL program. If visualization is the same as your result, then you should find the issue in GRBL program (or generator).

closed 2025-09-05 by github-actions[bot]