Denvi/Candle #644: Candle settings are stored incorrectly on Linux

Pauan, 2024-08-28:
**Versions**
 - Candle version: 1.1
 - Firmware version: 1.1

**PC info**
- OS: NixOS Unstable

**Describe the bug**

Candle currently stores the user settings in the directory where Candle is installed.

This is incorrect practice on Linux, and it [causes problems in distros such as NixOS](https://github.com/NixOS/nixpkgs/pull/134605).

Because of this, the settings are *never* saved, and so every time Candle is opened all of the settings are reset to the default.

The correct practice is to store user settings inside of the `$HOME/.config/candle` folder.

Or... alternatively, you could provide a setting so the user can specify where they want the settings stored. That would also fix this problem.

mar0x, 2024-08-28:
Hello,
One more possible solution: https://github.com/mar0x/Candle/commit/9bb522a6a5a57884bae009cc1982145397386b85

FaustXVI, 2025-04-16:
@Pauan I am having the same issue.
I tried to fix it with : 
```
                  cmakeFlags = [ 
                    "-DDEFAULT_SETTINGS_LOCATION=ON"
                  ];
```

But it still doesn't save the settings. Am I missing something? 
Thanks