Denvi/Candle #645: Rotary axis.

Hostaler, 2024-09-02:
Is it possible to schedule a rotating axis with candle?

JoanTheSpark, 2025-05-16:
https://www.reddit.com/r/hobbycnc/comments/xwbzul/comment/ir7gutd/?utm_source=share&utm_medium=web3x&utm_name=web3xcss&utm_term=1&utm_content=share_button

Hostaler, 2025-05-17:
> https://www.reddit.com/r/hobbycnc/comments/xwbzul/comment/ir7gutd/?utm_source=share&utm_medium=web3x&utm_name=web3xcss&utm_term=1&utm_content=share_button
> 
> > 
> 
> henpemaz • Hace 3 años Bricolaje de 5 ejes
> 
> He estado usando Candle con 4 ejes sin problemas; no hay nada intrínsecamente "incompatible". Simplemente no tendrás controles de desplazamiento fáciles y la vista previa ignorará el 4.º eje, pero ejecutará tus archivos sin problemas. Un emisor es un emisor; simplemente enviará el código G.
> 
> En la versión estable, hay cuatro comandos personalizados que puedes configurar y definir atajos de teclado para ellos, lo que permite mover el cuarto eje. En la rama experimental, puedes agregar tantos comandos personalizados como quieras.

Thank you very much.

closed 2025-05-17 by Hostaler