Denvi/Candle #646: Flip Z button directions in Candle

farsajik, 2024-09-07:
Hi.in my CNC in Z direction not spindle instead table moves up and down.so its opposite candle z buttons commands.everey time i have to think twice before pressing Z buttons).is it possible to flip Z buttons commands from candle settings?
Thank you

Hostaler, 2024-09-07:
Invierte la dirección de Z en los parámetros o cambia los cables del motor.

farsajik, 2024-09-07:
System is working correctly.but in my cnc configration  table goes up and down.spindle is static.i need reverse candle commands not cable

mar0x, 2024-09-07:
Hello @farsajik,
It looks you need to invert Z-axis direction. See GRBL Setting [$3 – Direction port invert](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#3--direction-port-invert-mask).

farsajik, 2024-09-07:
My grbl setting is correct.when runs gcode cnc mills, probes and engraves correctly.i only needs invert candle z axes commands

mar0x, 2024-09-08:
@farsajik Of course your GRBL settings are perfect! It would be good if you share it here.

I would suggest only to invert 3rd bit in [$3](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#3--direction-port-invert-mask) setting and most likely Candle will invert all movements along Z-axis.

Use $$ command in Candle console to show all GRBL settings. If you worry about privacy, share only current $3 value and I'll do my best to help you to calculate new value.

ondras12345, 2024-09-08:
> My grbl setting is correct.when runs gcode cnc mills, probes and engraves correctly.i only needs invert candle z axes commands

I think you need to stop thinking about the buttons in Candle as up/down, but rather Z+/Z-. When you press Z+ to make more space between the tool and the table, the table needs to go down. (Think about what the tool is doing relative to the table: it is going up.)

There is no way to invert the behavior of Z+/Z- buttons without changing the source code and recompiling:
https://github.com/Denvi/Candle/blob/e4c48e685b66f8610b066da10bb1be1e55eead9c/src/candle/frmmain.cpp#L1458-L1468

Hostaler, 2024-09-08:
Lo que tienes que hacer es explicar desde un principio que es la mesa lo que sube y baja, no la herramienta... Así no nos tienes a todos divagando porque a ti te sale de las narices.

mar0x, 2024-09-08:
> Lo que tienes que hacer es explicar desde un principio que es la mesa lo que sube y baja, no la herramienta... Así no nos tienes a todos divagando porque a ti te sale de las narices.

To be honest, @farsajik did this initially: "_.. not spindle instead table moves up and down_". The only think I have to guess is the GRBL program and probe command were adapted for this table movement. While Candle jog buttons are not.

farsajik, 2024-09-08:
$22=1
$21=1
$20=0
$102=400.000
$101=80.000
$100=80.000
$112=150.000
$111=3000.000
$110=3000.000
$122=10.000
$121=10.000
$120=10.000
$132=50.000
$131=300.000
$130=300.000
$10=1
$30=10000

Thank you for your help.I am using fluid nc.i need edit candle commands not grbl

farsajik, 2024-09-08:
If no way without recompilling candle, to solve this problem i created 2 macros MZ+ and MZ- 

> > Lo que tienes que hacer es explicar desde un principio que es la mesa lo que sube y baja, no la herramienta... Así no nos tienes a todos divagando porque a ti te sale de las narices.
> 
> To be honest, @farsajik did this initially: "_.. not spindle instead table moves up and down_". The only thing I have to guess is the GCode program and probe command were adapted for this table movement. While Candle jog buttons are not.
> 
> @Hostaler, пишите на английском, так удобнее для всех.

> > Lo que tienes que hacer es explicar desde un principio que es la mesa lo que sube y baja, no la herramienta... Así no nos tienes a todos divagando porque a ti te sale de las narices.
> 
> To be honest, @farsajik did this initially: "_.. not spindle instead table moves up and down_". The only thing I have to guess is the GCode program and probe command were adapted for this table movement. While Candle jog buttons are not.
> 
> @Hostaler, пишите на английском, так удобнее для всех.

Mar0x thank you.you understand correctly.exactly problem is candle jog buttons.there should be a setting to flip direction for my case.

I am working on small cnc for pcb milling.it has H-BOT kinematics with moving table

For normal kinematic when we press Z- or gcode commands tool goes down.distance between tool and table decreases.

In my case step motor moves table up and down.so to decrease distance between tool and table table must go up.i adopted motor to go correct direction when gcode commands.but candle job button inverse now.because my cnc has not srandart kinematics

farsajik, 2024-09-08:
[image]

Some think like this

mar0x, 2024-09-08:
@farsajik I still think your $3 GRBL setting is incorrect (you are not shared it [here](https://github.com/Denvi/Candle/issues/646#issuecomment-2336649518)) and this is the root of your issue. Positive Z+ movement should increase distance between tool and table, negative Z- movement should decrease it (just as @ondras12345 wrote [here](https://github.com/Denvi/Candle/issues/646#issuecomment-2336596322) ).

My CNC 3018 Pro also has moving table along Y axis, but GRBL direction setting is: Y+ moves the table to the _front_ and relative to detail, this makes the tool moves to the _back_ of the machine.

In other words, GCode program should control _the tool_ (not _the table_).

closed 2024-09-10 by farsajik